ROBOTIS-GIT/turtlebot3 #278: Question on the config file for cartographer

jbhorel, 2018-07-30:
Hello,

I have been working with a Turtlebot 3 waffle PI and cartographer for some months and I saw you changed two parameters in the config file : *imu_sampling_ratio* and *odometry_sampling_ratio*, from 1 to 0.1.

I ran the offline node with 5 different bags and several values of the *odometry_sampling_ratio*, no value was better than the others, it looked like one random value just fit well with one bag. I also didn't find a more theoretical explanation, so do you have a particular reason for this modifcation ?

Thanks in advance.

routiful, 2018-08-02:
Hi :)

This updates is to solve issue about time error. #228 #101 
We tested several times to figure out when it occurs and why.
The imu and odometry data was sent to OpenCR(Microcontroller). 
But this data is sometimes not synchronized due to latency of communication.
So I changed `sampling_ratio` to escape this problem.

jbhorel, 2018-08-02:
Hello,

Thanks for the answer, I understand why you decrease the sampling, I had the same issue with the time and imu in #228 so I stopped using it (I set TRAJECTORY_BUILDER_2D.use_imu_data to false) and I didn't see the difference. Thanks for resolving the issue, I will try with the imu by now.

Have you seen any differences in the quality of the maps with this ratio + imu data ?

routiful, 2018-08-03:
No, I can't find any differences the quality of the maps.

jbhorel, 2018-08-03:
Ok, many thanks for your answers, I close the issue as I have no questions left for the moment :).

closed 2018-08-03 by jbhorel

routiful, 2018-08-05:
Good, If you have any further questions, please feel free to open issue.

pitosalas, 2018-08-16:
I noticed on the web site that you recommend using cartographer 0.2 and not 0.3 with Kinetic. Is that still the recommendation?

routiful, 2018-08-17:
Yes, you are supposed to use cartographer 0.2.