ROBOTIS-GIT/turtlebot3 #279: Problem about Slam in Rviz

space03, 2018-07-31:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ v] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ v] etc (PLEASE, WRITE DOWN YOUR SBC HERE)JetsonTX2

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [v ] etc (PLEASE, WRITE DOWN YOUR OS)Ubuntu 16.04

4. Which OS you installed in Remote PC?

    - [ v] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

[Remote PC] roscore
[SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
[Remote PC] roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
[Remote PC] roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

7. Copy and Paste your error message on terminal
[image]

  
8. Please, describe detailedly what difficulty you are in 

when i run Slam, above error occured.
slam operation is working but i cannot know where the robot is in the map of rviz.
Please give me some help. 
Thank you.

routiful, 2018-08-02:
Hi :)

Before you launch slam, you are supposed to set TURTLEBOT3_MODEL parameter

```
$ export TURTLEBOT3_MODEL=waffle
```

If it isn't solved yet, please make sure the condition of topic communication by openning rqt

```
$ rqt
```

space03, 2018-08-03:
Thank you for your reply.

from your previous replies, I solved this problem by timesync.
http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

I'm using remote pc and turtlebot3 by connecting wireless router which is not online. So turtlebot and pc' time is not synchronized. After connecting Internet both pc and turtlebot or one by one,  timesync makes slam work well.

Thank you!

routiful, 2018-08-05:
Good, If you have any further questions, please feel free to open issue.

closed 2018-08-07 by robotpilot