ROBOTIS-GIT/turtlebot3 #283: cannot use turtlebot3_teleop launch.

idminerwilliam, 2018-08-02:
Hello,
I already updated software & openCR firmware to 1.3.1.

Button push test is OK.
Bringup is OK.
Use RC100 control is OK.

But when i roslaunch turtlebot3_teleop.launch, the turtlebot3 doesn't work.

what setting need modify ??
[image]
[image]

By the way, sometime when I bringup the romote.
it might show "Failed to get param: timeout expired" error
but I already ntpdate. time is the same.

Thx.

routiful, 2018-08-02:
Hi :)

Would you try to use `rostopic pub` commands?
```
rostopic pub /cmd_vel geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0" 
```

And if you set namespace on your turtlebot3, you are supposed to set `ROS NAMESPACE` like below link
http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#5-teleoperation

Furthermore, `Failed to get param: timeout expired` error is sometimes occurred due to timeout between OpenCR and Raspberry Pi. But this problem is simply solved by reboot OpenCR.

idminerwilliam, 2018-08-03:
Well.
I try use pub commands, still not working.
And I didn't set my turtlebot3 namespace @@. I just install raspbian on turtlebot3 and update turtlebot3 & turtlebot3_msg package.
This things happened when I update to 1.3.1 firmware.
Is the pi not receive my PC's /cmd_vel msg ?
I cannot see the /cmd_vel message between turtlebot3_teleop_keyboard and turtlebot3_core in rqt_graph.

routiful, 2018-08-05:
The v1.3.1 is OpenCR version.
Please check turtlebot3 software and firmware version to use below command.

```
$ rostopic echo /version_info
```

Can you show me your topic monitor?

```
$ rostopic list
```

or

```
$ rqt
```

idminerwilliam, 2018-08-06:
Hey, routiful

This is my /version_info :
---
hardware: "2017.05.30"
firmware: "1.2.2"
software: "1.1.0"
---
and rostopic list :
[image]

routiful, 2018-08-07:
I don't understand. But let's try more

1. Re-upload firmware

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#how-to-update-firmware

2. Check topic echo 
- Publish data

```
$ rostopic pub /cmd_vel geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0"
```
- Check it

```
$ rostopic echo /cmd_vel
```

robotpilot, 2018-08-13:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-13 by robotpilot

alsora, 2018-10-08:
Hi, I'm having a very similar issue. When I run 

`roslaunch turtlebot3_bringup turtlebot3_core.launch`

The USER1 led light starts flashing orange.

This is my `version_info` 
```
hardware: "2017.05.30"
firmware: "1.2.3"
software: "1.1.0"
```

I'm able to subscribe and receive data from all the topics, but when I try to teleoperate the robot it doesn't move.
I tried both with the teleoperation node, as well as publishing `geometry_msgs/Twist` on the `cmd_vel` topic.