ROBOTIS-GIT/turtlebot3 #286: Maximum velocities and related questions

pitosalas, 2018-08-06:
I am writing a new custom teleop application. I am using a turtlebot3 burger. 

I have read here and there that you have to be careful about what motion commands you give the robot because you might harm it. So I have some questions:

1) How fast is it ok to send cmd_vel topics
2) Should I stop sending them when a specific max speed is reached? What is that speed?
3) How about changing velocities: If I sent cmd_vel: x = 0.22 may I send one for cmd_vel x = 0 right after or do I need to step down gradually? How gradually?

Can you give me some guidance?

routiful, 2018-08-07:
Hi :)

You can find [limit velocity](http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#velocity-limit) of the Dynamixel which used in TurtleBot3 and [Max velocity of TurtleBot3](http://emanual.robotis.com/docs/en/platform/turtlebot3/specifications/#hardware-specifications)

So, even you want to set high speed to TurtleBot3, you can't do that due to limitation of the actuator.
I attached our node to control velocity of TurtleBot3. You can find simple way to make profile.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_teleop/nodes/turtlebot3_teleop_key#L77

pitosalas, 2018-08-07:
@routiful Thanks! I have seen the teleop_key code many times. It is careful to gradually change the velocity (make_simple_profile). This implies to me that your code is concerned about sending cmd_vel messages that change the velocity too quickly. Can you look again at my question (3) and answer it specifically?

routiful, 2018-08-07:
Then I recommend `MinimumJerk` method.
It might be help you.

http://courses.shadmehrlab.org/Shortcourse/minimumjerk.pdf

pitosalas, 2018-08-08:
Interesting... and is that algorithm built into the dynamixel or openCR or should I put it in my ros node to be safe?

closed 2018-08-08 by pitosalas

routiful, 2018-08-09:
Of course, the Dynamixel provides profile to smooth motion.
http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-velocity

routiful, 2018-08-09:
If you have any further question, please feel free to open issue.