ROBOTIS-GIT/turtlebot3 #287: Turtlebot 3 angular velocity commands sign change in real world

varunvp, 2018-08-07:
ISSUE TEMPLATE ver. 0.2.0

1. - [ ] Burger

2. - [ ] Raspberry Pi 3

3. - [ ] Raspbian

4. - [ ] Ubuntu 14.04 LTS

5. Write down software version and firmware version
    - Software version: [1.0.0]
    - Firmware version: [1.2.0]
 6. Write down the commands you used in order

    -roslaunch turtlebot3_bringup turtlebot3_robot.launch

I've noticed that I have to change the signs of the angular velocity commands whenever I switch between simulations, and the turtlebot. Is this a bug?

kijongGil, 2018-08-07:
Hi @varunvp :)
Could you tell me more about you said 'switch between simulations and the turtlebot'?

varunvp, 2018-08-07:
I meant switching between turtlebot simulations on gazebo, and the real life turtlebot.

kijongGil, 2018-08-07:
Could you check your DXL's ID? Letf DXL is 1, right DXL is 2.
Thanks,
Gilbert.

closed 2018-08-15 by robotpilot