ROBOTIS-GIT/turtlebot3 #289: Sudden velocity changes

pitosalas, 2018-08-08:
If I send a series of cmd_vel with x = 0.15 and immediately follow it with a cmd_vel with x = 0, with that harm the robot in any way?

routiful, 2018-08-09:
You are right. So, we provide additional node including simple profile.
https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_teleop

pitosalas, 2018-08-09:
@routiful @robotpilot you’re confusing me. I asked a yes/no question and you say “you are right”. So you mean that yes, it will harm the robot? Could you say more than one sentence? Also in another question below I think you said that the opencr has the profile built in which implied that the opencr logic would smooth things out. Please say more than one sentence?

routiful, 2018-08-09:
Sorry, my answer is YES. If Dynamixels are received signal that includes suddenly changes velocity, it is possible to damage to gears in Dynamixels.

Profile is not included in OpenCR. It only is included in `turtlebot3_teleop` node.

pitosalas, 2018-08-09:
(Sorry but I am one of those guys that really wants to understand exactly and then double check my understanding by asking more questions.)

@routiful Thanks! Except look at your answer in https://github.com/ROBOTIS-GIT/turtlebot3/issues/286 where you said that the OPenCR did provide the profile.

routiful, 2018-08-09:
No, I said `DYNAMIXEL` provides profile.

routiful, 2018-08-10:
I'm sorry that I can't give you exact answer you want. If you have any further question when you are proceeding it, please feel free to open issue.