ROBOTIS-GIT/turtlebot3 #290: Slam examples in Turtlebot3 Tutorial

pitosalas, 2018-08-10:
I am running the Slam and Navigation examples in the tutorial. It shows making the map first and then navigating on the map. How do I do both at the same time, that is, navigate over the known part of the map while expanding the map elsewhere?

robotpilot, 2018-08-10:
Hi @pitosalas,
Navigation uses the map file already created. So you can create a map using the SLAM package, and then use the navigation package using the created map.

I've set it up in the cartographer example file, not gmapping, to use both. Please refer to the explanation below.
```
roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer
```
http://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#run-slam-nodes

pitosalas, 2018-08-10:
@pyo thanks. I don’t understand the differnce in the two examples,at least from the reference you showed. Both use 

    roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer

The only difference that I see is the value for slam_methods, gmapping vs cartographer. What part enables the use of the map while navigating? Is that implicit in the cartographer library.

robotpilot, 2018-08-12:
You can also use navigation with the `move_base` in the launch file.

pitosalas, 2018-08-13:
@pyo could you answer in a little more detail?

robotpilot, 2018-08-15:
https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_slam/launch/turtlebot3_cartographer.launch#L22-L25

If you look at the `turtlebot3_cartographer.launch` file as above, you can navigate by running `move_base` as well as the cartographer node.

closed 2018-08-16 by pitosalas