ROBOTIS-GIT/turtlebot3 #291: Why include gazebo.xacro files into the urdf.xacro files

stevensu1838, 2018-08-12:
Hi,
May I please ask why do I have to include relevant gazebo.xacro files into the urdf.xacro files in the beginning? Do Gazebo and Rviz use this same urdf.xacro file? Or we only include the gazebo.xacro files for the use of robot visualization in Gazebo? Thanks.
Below is just an example of including a relevant gazebo.xacro file into the urdf.xacro file
<xacro:include filename="$(find turtlebot3_description)/urdf/turtlebot3_waffle.gazebo.xacro"/>

routiful, 2018-08-13:
HI :)

Rviz and Gazebo need to `robot_model` to represent robot in their environments.(Ref: http://wiki.ros.org/robot_model)

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_bringup/launch/includes/description.launch.xml

However, because Gazebo is 3d simulator, it needs additional information about inertia and color.
So this information has to be included in gazebo file by <gazebo> tag.

In summary, if you launched Rviz, you would not use gazebo file. 
But if you launched Gazebo, it needs

closed 2018-08-14 by stevensu1838

routiful, 2018-08-16:
If you have any further questions, please feel free to open issue :)