ROBOTIS-GIT/turtlebot3 #292: Cheapest wireless joystick recommended?

pitosalas, 2018-08-13:
I see many different "ps3 joysticks" on Amazon. Can you recommend one that will work easily and well?

robotpilot, 2018-08-13:
It is recommended to use genuine products.

pitosalas, 2018-08-14:
@pyo that's not a very helpful answer. I wasn't asking about genuine products (I don't even really know what they are.) What I was asking was what kind of joystick will work nicely with TB3 because I am buying one (I don't have a PS3 one lying around.) There are all kinds of them there - with lots of buttons that are not going to be used. So I was asking for a basic joystick that will work but is not fancy and is economical. Are you willing to tell me what you use?

robotpilot, 2018-08-15:
@pitosalas 

In fact, we are using the Sony PS3 joystick like the video below.
https://youtu.be/Z4s18hlazb4?t=6s

Other common joystick is RC-100B which we sell. It is also included as a basic component in the TurtleBot3 Waffle Pi product. This controller is directly connected to the robot with Bluetooth equipment.
http://www.robotis-shop-en.com/?act=shop_en.goods_view&GS=1487&GC=GD0B0106
https://youtu.be/Z4s18hlazb4?t=29s

closed 2018-08-16 by pitosalas