ROBOTIS-GIT/turtlebot3 #293: Possible error in book on page 320

pitosalas, 2018-08-13:
If I understand correctly...

On page 320 it is stated that the turtlebot3_slam turtlebot3_slam.launch will not launch rviz, and that Rviz needs to be run in the next step. But looking at the launch file, it does run Rviz. Am I missing something?

robotpilot, 2018-08-13:
I am sorry to confuse you. At the time of writing, I had to run RViz separately. But many people said that this was uncomfortable. So we modified the code to integrate. I will modify the book later. Thank you.

robotpilot, 2018-08-15:
Frequent updates to document work are common. I'm trying to get rid of these things as possible, but it's not easy. We are considering a few minor modifications and will collect them and redistribute them in the third quarter. Thank you for letting me know. :)

robotpilot, 2018-08-15:
I'm going to close this until update the issue with more information.

closed 2018-08-15 by robotpilot