ROBOTIS-GIT/turtlebot3 #295: Possible error in book page 323

pitosalas, 2018-08-13:
This one is a little more involved. I have a burger, but as I was going to run topics out of a bag I assumed it was ok to pretend that I have a waffle.

I saved the downloaded bag in my ~ directory
I ran the tb3_remote.launch without problem

$ export TURTLEBOT3_MODEL=waffle
$ rosrun rviz rviz -d `rospack find turtlebot3_slam`/rviz/turtlebot3_slam.rviz

And then I got error:

$ roscd turtlebot3_slam/bag
$ rosbag play ./TB3_WAFFLE_SLAM.bag

[ INFO] [1534196625.075223331]: Opening /home/robot/TB3_WAFFLE_SLAM.bag
**[FATAL] [1534196625.075631789]: Error reading version line**

Is the bag I downloaded bad?

robotpilot, 2018-08-13:
Hi @pitosalas
Something is strange. I downloaded it as mentioned on page 323 and the file was 40kB. Originally it is around 60MB.
```
wget https://github.com/ROBOTIS-GIT/bags/blob/master/TB3_WAFFLE_SLAM.bag
```
So, the following command was available to download normally. It is recommended to try again with the following command.
```
wget https://raw.githubusercontent.com/ROBOTIS-GIT/bags/master/TB3_WAFFLE_SLAM.bag
```
I am sorry and thank you for finding the problem.

pitosalas, 2018-08-14:
Also it looks like it has moved, out of the turtlebot3 GitHub repo (which is what is referenced in the book) to a new repo called bags (https://github.com/ROBOTIS-GIT/bags/blob/master/TB3_WAFFLE_SLAM.bag). As far as I can tell there are no bags in the turtlebot3 repo.

robotpilot, 2018-08-15:
This is because it is easier to manage relatively large files, such as bag files, from the source code. We appreciate your feedback and will revise this in the next update of the book. Thank you. :+1:

closed 2018-08-15 by robotpilot