ROBOTIS-GIT/turtlebot3 #296: Turtlebot3 Weight Distribution

pitosalas, 2018-08-14:
As mentioned in answers.ros.com, my TB3 exhibits a "bucking" or "stuttering" behavior when moving backwards. I am investigating whether somehow the TB3 is getting contradictory cmd_vel messages as we think might be the case.

I have a specific question for Robotis tough: Would balance or center of gravity affect this? Or would the condition of the caster ball? The caster ball does move freely. But I notice that the caster ball side is a little bit lower than the wheel side. Is that normal?

robotpilot, 2018-08-15:
For differential-drive mobile, a caster ball is used. This position, height, and center of gravity of the whole are very important. We tested the casters slightly lower through various tests. This is a proven part, so no problem.

robotpilot, 2018-08-16:
I have never seen you. If you think it is a hardware problem, please have it repaired by your dealer.

pitosalas, 2018-08-16:
I don't know exactly what you mean :) I am giving you information that I thought would be useful in case others have these problems. And also looking for help diagnosing the behavior.

closed 2018-08-16 by pitosalas