ROBOTIS-GIT/turtlebot3 #297: Turtlebot 3 burger doesn't move in "turtlebot3_teleop_key" demo

0x004z, 2018-08-14:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.12.13]
    - Firmware version: [1.2.2]
 
6. Write down the commands you used in order
    [Remote PC]
    $ roscore

    [turtlebot]
    $ roslaunch turtlebot3_bringup turtlebot3_robot.launch

    [Remote PC]
    $ export TURTLEBOT3_MODEL=burger
    $ roslaunch turtlebot3_bringup turtlebot3_remote.launch
    $ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

 
7. Copy and Paste your error message on terminal

    no error on terminal
  
8. Please, describe detailedly what difficulty you are in 

    I am pressing the "w, x, etc" keys but the turtlebot is not moving.  When pressing the user buttons turtlebot is moving so I suppose the assembly of the electronics is correct.
POWER LED: ON (green)
Arduino LED: blinking (green)
USER4 LED: ON (green)
USER1 LED: blinking(red)

``
SUMMARY
========

PARAMETERS
 * /model: burger
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    turtlebot3_teleop_keyboard (turtlebot3_teleop/turtlebot3_teleop_key)

ROS_MASTER_URI=http://192.168.2.63:11311

process[turtlebot3_teleop_keyboard-1]: started with pid [6247]

Control Your TurtleBot3!
---------------------------
Moving around:
        w
   a    s    d
        x

w/x : increase/decrease linear velocity (Burger : ~ 0.22, Waffle and Waffle Pi : ~ 0.26)
a/d : increase/decrease angular velocity (Burger : ~ 2.84, Waffle and Waffle Pi : ~ 1.82)

space key, s : force stop

CTRL-C to quit

currently:	linear vel 0.01	 angular vel 0.0 
currently:	linear vel 0.02	 angular vel 0.0 
currently:	linear vel 0.03	 angular vel 0.0 
currently:	linear vel 0.04	 angular vel 0.0 

``

routiful, 2018-08-16:
Hi :)

First, open topic monitor to check whether topics are working

http://emanual.robotis.com/docs/en/platform/turtlebot3/topic_monitor/#topic-monitor

Second, check internet(wifi) connection and time sync.

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#faq

routiful, 2018-08-17:
Good. If you have any further questions, please feel free to open issue.