ROBOTIS-GIT/turtlebot3 #300: Lidar spins as soon as Ubuntu Mate boots up on TB3 Burger

pitosalas, 2018-08-16:
I am setting up a new TB3 and unlike the other ones we have, this one has a couple of oddities, maybe related:

1. As soon as Ubuntu Mate is running, the lidar spins
1. The OpenCR continues with its led's flashing in a sequence
1. During booting of Mate, the very first message is "failed to start load kernel modules"

I am still installing ROS itself but everything seems totally normal other than those three weird behaviors. Do you know what they mean?

robotpilot, 2018-08-16:
1. Lida is designed to rotate at the same time as power is supplied.
2. The latest OpenCR only contains the firmware that flashes the LED at shipment. The user installs the firmware as described in the manual.
3.  It is a negligible part.

pitosalas, 2018-08-16:
Yes as I mentioned in my post, I think I had figured out 1 and 2. But 3 was still a concern and you've now answered that it is not a problem.

closed 2018-08-16 by pitosalas