ROBOTIS-GIT/turtlebot3 #303: LDS does not work.

jmaja, 2018-09-02:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - I have done all the installation as described on your Turtlebot3 Tutorial and everything works as presented in the website. But on "Load a TurtleBot3 on Rviz" - the LDS was rotating but I cannot see the point cloud on rviz as depicted on what was shown on the two figures. The robot model looks exactly the same in the picture but no red points that depicts the obstacle. Your input is highly appreciated.

jmaja, 2018-09-03:
> Hi @jmaja <https://github.com/jmaja> :)
> Could you tell me about what did you launchs?

Gilbert,

Good morning. This is the three things I did following the tutorial:

$ export TURTLEBOT3_MODEL=${TB3_MODEL}$ roslaunch turtlebot3_bringup
turtlebot3_remote.launch$ rosrun rviz rviz -d `rospack find
turtlebot3_description`/rviz/model.rviz

Joe

kijongGil, 2018-09-05:
I'm sorry for late reply.
First, Let's confirm your LDS. 

1. Run roscore in Remote PC
`roscoe`
 2. Bring up your turtlebot3.
`roslaunch turtlebot3_bringup turtlebot3_robot.launch`
3. echo scan topic.
`rostopic echo /scan`

If your LDS is working, you can see LDS data.

closed 2018-09-11 by robotpilot