ROBOTIS-GIT/turtlebot3 #304: ros-kinetic-map-server - install error - libsdl-image1.2-dev and libsdl1.2-dev dependency

senake, 2018-09-04:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?...Not applicable. I am currently stuck on ROS/Ubuntu PC install

    - [ ] Ubuntu MATE 16.04 or later
    - [X] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware -
    ...Not applicable. I am currently stuck on ROS/Ubuntu PC install

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
I followed the instructions in the Robotis Turtlebot3 manual here -  http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/

(A) Install Ubuntu on PC - ISSUE #1

Because the current version of Ubuntu is 18.0.4, I couldn't use the link in your user manual to install Ubuntu 16.0.4.  The download link in the Robotis manual on page (http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/) is https://www.ubuntu.com/download/alternative-downloads. The 16.0.4 link leads to this page http://cdimage.ubuntu.com/netboot/16.04/?_ga=2.105723779.1610410129.1535970811-665810113.1535820947 which has more links, not an ISO image as would be expected from the URL. The amd64 link which I used (as I was installing Ubuntu over a previous Windows 7) then led to this page - http://archive.ubuntu.com/ubuntu/dists/xenial-updates/main/installer-amd64/current/images/netboot/ - which just had directories and files, not and ISO image.

For the reason above, I used the following links instead:

a. Link to obtain the ISO image of Ubuntu 16.0.4 from http://releases.ubuntu.com/16.04/. The ISO image is at:

   http://releases.ubuntu.com/16.04/ubuntu-16.04.5-desktop-amd64.iso

b. Instructions to create a bootable USB:

https://tutorials.ubuntu.com/tutorial/tutorial-create-a-usb-stick-on-windows?_ga=2.157960632.304329474.1535820947-665810113.1535820947#3

(B) Install ROS

I then followed instructions 6.1.2 'Install ROS on remote PC' using the terminal commands below. This worked fine.

$ sudo apt-get update
$ sudo apt-get upgrade
$ wget https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros_kinetic.sh && chmod 755 ./install_ros_kinetic.sh && bash ./install_ros_kinetic.sh

(C) Install dependent ROS packages - ISSUE #2 

The command below installed most of the dependent packages - see state below - but it generated an installing ros-kinetic-map-server. This is the error upon which I an currently stuck:

$ sudo apt-get install ros-kinetic-joy ros-kinetic-teleop-twist-joy ros-kinetic-teleop-twist-keyboard ros-kinetic-laser-proc ros-kinetic-rgbd-launch ros-kinetic-depthimage-to-laserscan ros-kinetic-rosserial-arduino ros-kinetic-rosserial-python ros-kinetic-rosserial-server ros-kinetic-rosserial-client ros-kinetic-rosserial-msgs ros-kinetic-amcl ros-kinetic-map-server ros-kinetic-move-base ros-kinetic-urdf ros-kinetic-xacro ros-kinetic-compressed-image-transport ros-kinetic-rqt-image-view ros-kinetic-gmapping ros-kinetic-navigation ros-kinetic-interactive-markers

Reading package lists... Done Building dependency tree
Reading state information... Done ros-kinetic-compressed-image-transport is already the newest version (1.9.5-0xenial-20180809-161709-0800). ros-kinetic-compressed-image-transport set to manually installed. ros-kinetic-interactive-markers is already the newest version (1.11.4-0xenial-20180809-172616-0800). ros-kinetic-interactive-markers set to manually installed. ros-kinetic-rqt-image-view is already the newest version (0.4.13-0xenial-20180809-163145-0800). ros-kinetic-rqt-image-view set to manually installed. ros-kinetic-urdf is already the newest version (1.12.12-0xenial-20180809-153440-0800). ros-kinetic-urdf set to manually installed. ros-kinetic-xacro is already the newest version (1.11.3-0xenial-20180809-153500-0800). ros-kinetic-xacro set to manually installed.

**Some packages could not be installed. This may mean that you have requested an impossible situation or if you are using the unstable distribution that some required packages have not yet been created or been moved out of Incoming. The following information may help to resolve the situation:

The following packages have unmet dependencies. ros-kinetic-map-server : Depends: libsdl-image1.2-dev but it is not going to be installed Depends: libsdl1.2-dev but it is not going to be installed E: Unable to correct problems, you have held broken packages.**
 
7. Copy and Paste your error message on terminal

    - HERE
**Some packages could not be installed. This may mean that you have requested an impossible situation or if you are using the unstable distribution that some required packages have not yet been created or been moved out of Incoming. The following information may help to resolve the situation:

The following packages have unmet dependencies. ros-kinetic-map-server : Depends: libsdl-image1.2-dev but it is not going to be installed Depends: libsdl1.2-dev but it is not going to be installed E: Unable to correct problems, you have held broken packages.**

  
8. Please, describe detailedly what difficulty you are in 

    - HERE
************************************************************************************************
I cannot complete the ROS install on my PC to use with my new Turtlebot 3.
************************************************************************************************

See issue resolution efforts here on the Robotis forum. Your support engineer Kurt suggested logging an issue here on Github if his suggested steps didn't resolve the issue.

Kurt's suggested steps to check if I have all of the Ubuntu and ROS components & the output from my session are below. The Ubuntu and non dependent ROS packages all seem to be complete and up to date.

sa@sa-ThinkPad-X230:~$ sudo apt-get update
Hit:1 http://gb.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://packages.ros.org/ros/ubuntu xenial InRelease
Reading package lists... Done
sa@sa-ThinkPad-X230:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
sa@sa-ThinkPad-X230:~$ sudo apt-get dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
sa@sa-ThinkPad-X230:~$ sudo apt-get autoremove
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
sa@sa-ThinkPad-X230:~$ sudo apt-get install ros-kinetic-desktop-full
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-kinetic-desktop-full is already the newest version (1.3.2-0xenial-20180810-032811-0800).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
sa@sa-ThinkPad-X230:~$

kijongGil, 2018-09-05:
Hi @senake :)
Could you show me the result that came up with the following command?
`apt-cache showpkg liferea`

kijongGil, 2018-09-05:
I think that you could change  gb.archive.ubuntu.com to us.archive.ubuntu.com by typing
`sudo gedit /etc/apt/sources.list`

senake, 2018-09-05:
I edited the sources list and retried the ROS install command. The edit
command gave some errors - see below - but still opened the document in a
text editor. I changed the two ucommented entries from gb to us. I then
tried rerunning the ROS install command, but now get an additional errors
for other ROS packages - see below.

sa@sa-ThinkPad-X230:~$ sudo gedit /etc/apt/sources.list
[sudo] password for sen:

(gedit:3456): Gtk-WARNING **: Calling Inhibit failed:
GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name
org.gnome.SessionManager was not provided by any .service files

** (gedit:3456): WARNING **: Set document metadata failed: Setting
attribute metadata::gedit-spell-enabled not supported

** (gedit:3456): WARNING **: Set document metadata failed: Setting
attribute metadata::gedit-encoding not supported

** (gedit:3456): WARNING **: Set document metadata failed: Setting
attribute metadata::gedit-position not supported

sa@sa-ThinkPad-X230:~$ sudo apt-get install ros-kinetic-joy
ros-kinetic-teleop-twist-joy ros-kinetic-teleop-twist-keyboard
ros-kinetic-laser-proc ros-kinetic-rgbd-launch
ros-kinetic-depthimage-to-laserscan ros-kinetic-rosserial-arduino
ros-kinetic-rosserial-python ros-kinetic-rosserial-server
ros-kinetic-rosserial-client ros-kinetic-rosserial-msgs ros-kinetic-amcl
ros-kinetic-map-server ros-kinetic-move-base ros-kinetic-urdf
ros-kinetic-xacro ros-kinetic-compressed-image-transport
ros-kinetic-rqt-image-view ros-kinetic-gmapping ros-kinetic-navigation
ros-kinetic-interactive-markers
Reading package lists... Done
Building dependency tree
Reading state information... Done
ros-kinetic-compressed-image-transport is already the newest version
(1.9.5-0xenial-20180809-161709-0800).
ros-kinetic-compressed-image-transport set to manually installed.
ros-kinetic-interactive-markers is already the newest version
(1.11.4-0xenial-20180809-172616-0800).
ros-kinetic-interactive-markers set to manually installed.
ros-kinetic-rqt-image-view is already the newest version
(0.4.13-0xenial-20180809-163145-0800).
ros-kinetic-rqt-image-view set to manually installed.
ros-kinetic-urdf is already the newest version
(1.12.12-0xenial-20180809-153440-0800).
ros-kinetic-urdf set to manually installed.
ros-kinetic-xacro is already the newest version
(1.11.3-0xenial-20180809-153500-0800).
ros-kinetic-xacro set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 ros-kinetic-joy : Depends: joystick but it is not installable
 ros-kinetic-map-server : Depends: libsdl-image1.2 (>= 1.2.10) but it is
not installable
                          Depends: libbullet-dev but it is not installable
                          Depends: libsdl-image1.2-dev but it is not
installable
                          Depends: libsdl1.2-dev but it is not going to be
installed
 ros-kinetic-navigation : Depends: ros-kinetic-robot-pose-ekf but it is not
going to be installed
 ros-kinetic-rosserial-arduino : Depends: arduino-core but it is not
installable
E: Unable to correct problems, you have held broken packages.a

kijongGil, 2018-09-05:
Your mirrors in ubuntu sever was not changed. This will be help you. 
https://askubuntu.com/questions/104695/how-do-i-change-mirrors-in-ubuntu-server-from-regional-to-main

senake, 2018-09-05:
I changed the setting i the GUI - see screen shot. It was previously set to
'custom servers'. I retried the install command, but still get the original
error:

sa@sa-ThinkPad-X230:~$ sudo apt-get install ros-kinetic-joy
ros-kinetic-teleop-twist-joy ros-kinetic-teleop-twist-keyboard
ros-kinetic-laser-proc ros-kinetic-rgbd-launch
ros-kinetic-depthimage-to-laserscan ros-kinetic-rosserial-arduino
ros-kinetic-rosserial-python ros-kinetic-rosserial-server
ros-kinetic-rosserial-client ros-kinetic-rosserial-msgs ros-kinetic-amcl
ros-kinetic-map-server ros-kinetic-move-base ros-kinetic-urdf
ros-kinetic-xacro ros-kinetic-compressed-image-transport
ros-kinetic-rqt-image-view ros-kinetic-gmapping ros-kinetic-navigation
ros-kinetic-interactive-markers
[sudo] password for sen:
Reading package lists... Done
Building dependency tree
Reading state information... Done
ros-kinetic-compressed-image-transport is already the newest version
(1.9.5-0xenial-20180809-161709-0800).
ros-kinetic-compressed-image-transport set to manually installed.
ros-kinetic-interactive-markers is already the newest version
(1.11.4-0xenial-20180809-172616-0800).
ros-kinetic-interactive-markers set to manually installed.
ros-kinetic-rqt-image-view is already the newest version
(0.4.13-0xenial-20180809-163145-0800).
ros-kinetic-rqt-image-view set to manually installed.
ros-kinetic-urdf is already the newest version
(1.12.12-0xenial-20180809-153440-0800).
ros-kinetic-urdf set to manually installed.
ros-kinetic-xacro is already the newest version
(1.11.3-0xenial-20180809-153500-0800).
ros-kinetic-xacro set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 ros-kinetic-map-server : Depends: libsdl-image1.2-dev but it is not going
to be installed
                          Depends: libsdl1.2-dev but it is not going to be
installed
E: Unable to correct problems, you have held broken packages.
sa@sa-ThinkPad-X230:~$

kijongGil, 2018-09-05:
How about install ros according to manual installation?
http://wiki.ros.org/kinetic/Installation/Ubuntu

senake, 2018-09-05:
That seemed to work - see below.

Please can you let me know how I can check if map server and the other
packages installed correctly?

sa@sa-ThinkPad-X230:~$ sudo sh -c 'echo "deb
http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" >
/etc/apt/sources.list.d/ros-latest.list'
[sudo] password for sen:
sa@sa-ThinkPad-X230:~$ sudo apt-key adv --keyserver hkp://
ha.pool.sks-keyservers.net:80 --recv-key
421C365BD9FF1F717815A3895523BAEEB01FA116
Executing: /tmp/tmp.KlZ7pBJEQw/gpg.1.sh --keyserver
hkp://ha.pool.sks-keyservers.net:80
--recv-key
421C365BD9FF1F717815A3895523BAEEB01FA116
gpg: requesting key B01FA116 from hkp server ha.pool.sks-keyservers.net
gpg: key B01FA116: "ROS Builder <rosbuild@ros.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
sa@sa-ThinkPad-X230:~$ sudo apt-get update
Hit:1 http://us.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://packages.ros.org/ros/ubuntu xenial InRelease
Reading package lists... Done
sa@sa-ThinkPad-X230:~$ sudo apt-get install ros-kinetic-desktop-full
Reading package lists... Done
Building dependency tree
Reading state information... Done
ros-kinetic-desktop-full is already the newest version
(1.3.2-0xenial-20180810-032811-0800).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
sen@sen-ThinkPad-X230:~$ sudo rosdep init
ERROR: default sources list file already exists:
    /etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
sa@sa-ThinkPad-X230:~$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit
https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index
https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
updated cache in /home/sen/.ros/rosdep/sources.cache
sa@sa-ThinkPad-X230:~$ echo "source /opt/ros/kinetic/setup.bash" >>
~/.bashrc
sa@sa-ThinkPad-X230:~$ source ~/.bashrc

kijongGil, 2018-09-05:
You can check packages installed 
`apt list --installed`

senake, 2018-09-05:
I couldn't see map_server ad navigation in the installed ROS packages - see
below. Shall I run the previous install script (that failed) to add these?

ros-kinetic-actionlib/xenial,now 1.11.13-0xenial-20180809-155829-0800 amd64
[installed,automatic]
ros-kinetic-actionlib-msgs/xenial,now 1.12.6-0xenial-20180809-141505-0800
amd64 [installed,automatic]
ros-kinetic-actionlib-tutorials/xenial,now
0.1.10-0xenial-20180809-160646-0800 amd64 [installed,automatic]
ros-kinetic-angles/xenial,now 1.9.11-0xenial-20180809-141106-0800 amd64
[installed,automatic]
ros-kinetic-bond/xenial,now 1.8.1-0xenial-20180809-141537-0800 amd64
[installed,automatic]
ros-kinetic-bond-core/xenial,now 1.8.1-0xenial-20180809-150208-0800 amd64
[installed,automatic]
ros-kinetic-bondcpp/xenial,now 1.8.1-0xenial-20180809-145306-0800 amd64
[installed,automatic]
ros-kinetic-bondpy/xenial,now 1.8.1-0xenial-20180809-145637-0800 amd64
[installed,automatic]
ros-kinetic-camera-calibration/xenial,now
1.12.23-0xenial-20180809-200914-0800 amd64 [installed,automatic]
ros-kinetic-camera-calibration-parsers/xenial,now
1.11.13-0xenial-20180809-160317-0800 amd64 [installed,automatic]
ros-kinetic-camera-info-manager/xenial,now
1.11.13-0xenial-20180809-161647-0800 amd64 [installed,automatic]
ros-kinetic-catkin/xenial,now 0.7.14-0xenial-20180809-132632-0800 amd64
[installed,automatic]
ros-kinetic-class-loader/xenial,now 0.3.9-0xenial-20180809-135528-0800
amd64 [installed,automatic]
ros-kinetic-cmake-modules/xenial,now 0.4.1-0xenial-20180809-135030-0800
amd64 [installed,automatic]
ros-kinetic-collada-parser/xenial,now 1.12.12-0xenial-20180809-154235-0800
amd64 [installed,automatic]
ros-kinetic-collada-urdf/xenial,now 1.12.12-0xenial-20180809-155009-0800
amd64 [installed,automatic]
ros-kinetic-common-msgs/xenial,now 1.12.6-0xenial-20180809-201132-0800
amd64 [installed,automatic]
ros-kinetic-common-tutorials/xenial,now 0.1.10-0xenial-20180809-162433-0800
amd64 [installed,automatic]
ros-kinetic-compressed-depth-image-transport/xenial,now
1.9.5-0xenial-20180809-161724-0800 amd64 [installed,automatic]
ros-kinetic-compressed-image-transport/xenial,now
1.9.5-0xenial-20180809-161709-0800 amd64 [installed,automatic]
ros-kinetic-control-msgs/xenial,now 1.4.0-0xenial-20180809-143428-0800
amd64 [installed,automatic]
ros-kinetic-cpp-common/xenial,now 0.6.9-0xenial-20180809-134002-0800 amd64
[installed,automatic]
ros-kinetic-cv-bridge/xenial,now 1.12.8-0xenial-20180809-155803-0800 amd64
[installed,automatic]
ros-kinetic-depth-image-proc/xenial,now
1.12.23-0xenial-20180809-193515-0800 amd64 [installed,automatic]
ros-kinetic-desktop/xenial,now 1.3.2-0xenial-20180810-032708-0800 amd64
[installed,automatic]
ros-kinetic-desktop-full/xenial,now 1.3.2-0xenial-20180810-032811-0800
amd64 [installed]
ros-kinetic-diagnostic-aggregator/xenial,now
1.9.3-0xenial-20180809-152946-0800 amd64 [installed,automatic]
ros-kinetic-diagnostic-analysis/xenial,now
1.9.3-0xenial-20180809-154703-0800 amd64 [installed,automatic]
ros-kinetic-diagnostic-common-diagnostics/xenial,now
1.9.3-0xenial-20180809-170814-0800 amd64 [installed,automatic]
ros-kinetic-diagnostic-msgs/xenial,now 1.12.6-0xenial-20180809-141702-0800
amd64 [installed,automatic]
ros-kinetic-diagnostic-updater/xenial,now
1.9.3-0xenial-20180809-153037-0800 amd64 [installed,automatic]
ros-kinetic-diagnostics/xenial,now 1.9.3-0xenial-20180809-172326-0800 amd64
[installed,automatic]
ros-kinetic-dynamic-reconfigure/xenial,now
1.5.49-0xenial-20180809-160102-0800 amd64 [installed,automatic]
ros-kinetic-eigen-conversions/xenial,now
[... 366 lines cut ...]
[installed,automatic]
ros-kinetic-turtlesim/xenial,now 0.7.1-0xenial-20180809-145406-0800 amd64
[installed,automatic]
ros-kinetic-urdf/xenial,now 1.12.12-0xenial-20180809-153440-0800 amd64
[installed,automatic]
ros-kinetic-urdf-parser-plugin/xenial,now
1.12.12-0xenial-20180809-141354-0800 amd64 [installed,automatic]
ros-kinetic-urdf-tutorial/xenial,now 0.3.0-1xenial-20180809-210753-0800
amd64 [installed,automatic]
ros-kinetic-vision-opencv/xenial,now 1.12.8-0xenial-20180809-201058-0800
amd64 [installed,automatic]
ros-kinetic-visualization-marker-tutorials/xenial,now
0.10.3-0xenial-20180809-145436-0800 amd64 [installed,automatic]
ros-kinetic-visualization-msgs/xenial,now
1.12.6-0xenial-20180809-142930-0800 amd64 [installed,automatic]
ros-kinetic-visualization-tutorials/xenial,now
0.10.3-0xenial-20180809-214616-0800 amd64 [installed,automatic]
ros-kinetic-viz/xenial,now 1.3.2-0xenial-20180810-032444-0800 amd64
[installed,automatic]
ros-kinetic-webkit-dependency/xenial,now 1.1.0-0xenial-20180809-143944-0800
amd64 [installed,automatic]
ros-kinetic-xacro/xenial,now 1.11.3-0xenial-20180809-153500-0800 amd64
[installed,automatic]
ros-kinetic-xmlrpcpp/xenial,now 1.12.13-0xenial-20180809-135412-0800 amd64
[installed,automatic]

senake, 2018-09-05:
Actually, I just tried it and it did fail with the same error....

sa@sa-ThinkPad-X230:~$ sudo apt-get install ros-kinetic-joy
ros-kinetic-teleop-twist-joy ros-kinetic-teleop-twist-keyboard
ros-kinetic-laser-proc ros-kinetic-rgbd-launch
ros-kinetic-depthimage-to-laserscan ros-kinetic-rosserial-arduino
ros-kinetic-rosserial-python ros-kinetic-rosserial-server
ros-kinetic-rosserial-client ros-kinetic-rosserial-msgs ros-kinetic-amcl
ros-kinetic-map-server ros-kinetic-move-base ros-kinetic-urdf
ros-kinetic-xacro ros-kinetic-compressed-image-transport
ros-kinetic-rqt-image-view ros-kinetic-gmapping ros-kinetic-navigation
ros-kinetic-interactive-markers
[sudo] password for sen:
Reading package lists... Done
Building dependency tree
Reading state information... Done
ros-kinetic-compressed-image-transport is already the newest version
(1.9.5-0xenial-20180809-161709-0800).
ros-kinetic-compressed-image-transport set to manually installed.
ros-kinetic-interactive-markers is already the newest version
(1.11.4-0xenial-20180809-172616-0800).
ros-kinetic-interactive-markers set to manually installed.
ros-kinetic-rqt-image-view is already the newest version
(0.4.13-0xenial-20180809-163145-0800).
ros-kinetic-rqt-image-view set to manually installed.
ros-kinetic-urdf is already the newest version
(1.12.12-0xenial-20180809-153440-0800).
ros-kinetic-urdf set to manually installed.
ros-kinetic-xacro is already the newest version
(1.11.3-0xenial-20180809-153500-0800).
ros-kinetic-xacro set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 ros-kinetic-map-server : Depends: libsdl-image1.2-dev but it is not going
to be installed
                          Depends: libsdl1.2-dev but it is not going to be
installed
E: Unable to correct problems, you have held broken packages.
sa@sa-ThinkPad-X230:~$

kijongGil, 2018-09-06:
I think you should format your PC and try again from making ubuntu booting USB.
After reinstalling ubuntu, change the ubuntu sever and install ros.

senake, 2018-09-06:
Hi Gilbert,

Okay, I will give that a try.

Thanks

Senake

senake, 2018-09-06:
Hi Gilbert,

As suggested, I reinstalled Ubuntu 16.0.4. with the full disk format option
and then installed ROS. This time, the install went a lot further, but
errored on the last step in the Robotis eManual PC setup - see error below.
It said that catkin_make was not installed and suggested the command for
it's installation. I ran this but a dependent file was missing.

A text file containing the full ROS install output is attached.  I searched
it for "python_catkin_pkg" but couldn't see any errors in the install
process. I did, however notice - see extract below. There were, however, a
couple of 'pt_thread_create' not found messages in the catkin make part of
the output - see highlighted extract below. Could these be the cause of the
issue?

Please can you let me know what to try next.

Thanks

Senake

*ERROR*

sa@sa-ThinkPad-X230:~/catkin_ws/src$ cd ~/catkin_ws && catkin_make
The program 'catkin_make' is currently not installed. You can install it by
typing:
sudo apt install catkin

sa@sa-ThinkPad-X230:~/catkin_ws$ sudo apt install catkin
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.

* catkin : Depends: python-catkin-pkg but it is not going to be installedE:
Unable to correct problems, you have held broken
packages.sa@sa-ThinkPad-X230:~/catkin_ws*

*EXTRACT FROM ROS INSTALL OUTPUT:*

[Make the catkin workspace and test the catkin_make]
Creating symlink "/home/sa/catkin_ws/src/CMakeLists.txt" pointing to
"/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
Base path: /home/sa/catkin_ws
Source space: /home/sa/catkin_ws/src
Build space: /home/sa/catkin_ws/build
Devel space: /home/sa/catkin_ws/devel
Install space: /home/sa/catkin_ws/install
####
#### Running command: "cmake /home/sa/catkin_ws/src
-DCATKIN_DEVEL_PREFIX=/home/sa/catkin_ws/devel
-DCMAKE_INSTALL_PREFIX=/home/sa/catkin_ws/install -G Unix Makefiles" in
"/home/sa/catkin_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/sa/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/sa/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
*-- Looking for pthread_create - not found*
-- Looking for pthread_create in pthreads
*-- Looking for pthread_create in pthreads - not found*
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sa/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/sa/catkin_ws/build"
####
[Set the ROS evironment]
[Complete!!!]

kijongGil, 2018-09-07:
You have not installed ROS completely.
How about install another linux version? I recommend installing linux mint(https://linuxmint.com/release.php?id=31).

senake, 2018-09-07:
Hi Gilbert,

Is it not possible to tell what part of the ROS install failed and why from
the log that I attached to my last email?

If possible, I would rather stick with Ubuntu 16.0.4 as it is what most
other ROS users are using.

Thanks

Senake

kijongGil, 2018-09-07:
I don't understand your situation. I think there are two causes.
One is your network. Another is ubuntu version.
So, I recommend installing linux mint 18. This is no different from ubuntu 16.04. I am also using linux mint 18.

senake, 2018-09-07:
Hi Gilbert,

Apologies, I didn't realise that ROS had been developed for the Linux Mint
distro too.

Please can you let me know which Linux Mint edition to chose on the link
that you provided.

Thanks

Senake

senake, 2018-09-08:
Hi Gilbert,

I tried the install on Linux Mint, but it failed with the same error as o
Ubuntu 16.0.4. I then tried 16.0.4 32 bit in case my hardware was a
limiting factor, but my laptop (an old X230 with a 225Gb SSD, 16GB RAM and
Intel i5-3320M CPU) ran out of hard disk space when installing ROS, so
maybe the 32bit OD did not address all of my hard disk.

I reinstalled Ubuntu 16.0.4 64 bit and then reran the ROS install manually
using the instructions on your/the Turtlebot3 manual link (
http://wiki.ros.org/kinetic/Installation/Ubuntu). After resuming and
installing the ROS dependent packages as per the eMaual (
http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/), *I found
that the catkin_ws directory had still not been created i my /home/sa
directory so I followed these steps -
http://wiki.ros.org/catkin/Tutorials/create_a_workspace
<http://wiki.ros.org/catkin/Tutorials/create_a_workspace> - to manually
create the catkin workspace. It has ow been created and is in the path...*

*sa@sa-ThinkPad-X230:~/catkin_ws$ echo
$ROS_PACKAGE_PATH/home/sa/catkin_ws/src:/opt/ros/kinetic/share*

*Do you think that this install is now okay?*

Please let me know.

Thanks

Senake

senake, 2018-09-08:
Actually, in trying to implement the next step in the manual - network
configuration, I noticed that my bash.rc file has a number of omitted lines
compared to the one shown in the manual - see below.

Please can you suggest what I can try now?

[image: Screenshot from 2018-09-08 09-57-20.png]

[image: Screenshot from 2018-09-08 10-05-00.png]
[image: Screenshot from 2018-09-08 09-57-00.png]

senake, 2018-09-10:
Issue resolved (I think)...

After running 'source /opt/ros/kinetic/setup.bash' as recommended in this
forum post -
https://answers.ros.org/question/258444/sudo-apt-install-catkin-not-working/
- the catkin make command builds everything without errors. I am not sure
why, but it should probably be added to the Turtlebot3 eManual.

kijongGil, 2018-09-10:
I'm sorry for late reply. I'm glad you installed ROS. This command 'source /opt/ros/kinetic/setup.bash' is include in install_ros_kinetic.sh. But, sometimes there times when this does not apply, I'm sorry I did not get noticed.

I'm going to close this issue. If you have a issue, you can reopen it again at any time.
Thanks.

closed 2018-09-10 by kijongGil

kijongGil, 2018-09-12:
Hi, again. 
I'm sorry, your new question does not fit into this issue, could you open an issue in a new title?

senake, 2018-09-12:
Hi Gilbert,

I have asked it here just now -
https://answers.ros.org/question/303106/hls_lfcd_lds_driverhlds_laser_publisher-error-on-roslaunch-turtlebot3_bringup-turtlebot3_robotlaunch/

senake, 2018-09-23:
Hi Gilbert,

No one has provided an answer to this question yet -
https://answers.ros.org/question/303106/hls_lfcd_lds_driverhlds_laser_publisher-error-on-roslaunch-turtlebot3_bringup-turtlebot3_robotlaunch/
.

1. Please can you let me know what to do?

2. My order with Robosavvy was actually for a TurtleBot3 Pi with an R200
Realsense developer kit that includes an Intel Atom based UP SBC. This
question relates to getting the TurtleBot3 up and running with the Pi. I
have also looked at getting the UP board working, bu there are no
instructions in the TurtleBot eManual for the set up of this SBC. I have
created a seperate issue for that here -
https://github.com/ROBOTIS-GIT/turtlebot3/issues/321

Thanks

Senake

kishuinvain1, 2019-09-22:
**sudo aptitude install ros-kinetic-map-server**
will definitely resolve the dependency issues.