ROBOTIS-GIT/turtlebot3 #305: turtlebot3 can't walk along straight lines.

lhv5, 2018-09-07:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
            
       Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

       Raspberry Pi 3    

3. Which OS you installed in SBC?
   
       Raspbian   

4. Which OS you installed in Remote PC?

       Ubuntu 16.04 LTS    

5. Write down software version and firmware version

        Software version: [x.x.x]
        Firmware version: [1.3.1]
   
6. Please, describe detailedly what difficulty you are in 

       When the robot moves forward, there will be a deviation from the straight line. How can I solve this problem?
       I think of two ways to solve this problem, calibration wheel diameter, or adjust the motor parameters. But I don't know exactly how to do, and I didn't find where to adjust, please tell me the specific realization method.

lhv5, 2018-09-07:
Hi @kijongGil 
    When turtlebot3 moves forward to 5M, there are about 20CM deviation. I use remote control(rc100) to control the turtlebot3 moving.     
    I read the part of the code of the openCR, but I don't know where to  adjust  the motor parameters, and do not know which parameter should be changed. Could you explain it in detail?   Thanks

kijongGil, 2018-09-07:
There are codes you want in https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/turtlebot3
If you want to modified OpenCR firmware, this will be help you.
http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-firmware-upload-for-tb3

However, in my opinion, it seems to be a Dynamixel problem or a driving environment problem. How about check your dynamixel? and did not the slip occur while the turtlebot3 was moving?

lhv5, 2018-09-11:
@kijongGil     Thank you for your advice. I found this problem caused by skid, because the floor was carpeted. But I wonder how turtlebot3 adjusts itself when it doesn't follow a straight line. What's the principle? By adjusting the PID parameter?

kijongGil, 2018-09-13:
Turtlebot3 has an imu sensor. how about use imu sensor?
You can get an orientation from imu data. Then control to PID parameter from this.

closed 2018-09-20 by kijongGil