ROBOTIS-GIT/turtlebot3 #306: Installing ROS2 for Turtlebot 3 on MacOS

nwesthoff, 2018-09-10:
1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [x] MacOS High Sierra

5. Write down software version and firmware version

    - Software version: x
    - Firmware version: 1.2.0
 
6. Write down the commands you used in order

1. First went through the full ROS2 on OS X installation [guide](https://github.com/ros2/ros2/wiki/OSX-Development-Setup)
2. added `source ~/ros2_ws/install/local_setup.bash` to bash profile and sourced it
3. created `turtlebot3_ws` and cloned git repos
4. ran `cd ~/turtlebot3_ws && colcon build`

7. Copy and Paste your error message on terminal

```
Starting >>> cartographer
Starting >>> cartographer_ros_msgs
Starting >>> pcl_conversions
Starting >>> turtlebot3_description
Starting >>> turtlebot3_msgs                                
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_teleop [5.24s]                            
[Processing: cartographer, cartographer_ros_msgs, pcl_conversions, turtlebot3_description, turtlebot3_msgs]
[Processing: cartographer, cartographer_ros_msgs, pcl_conversions, turtlebot3_description, turtlebot3_msgs]
--- stderr: cartographer_ros_msgs                                           
CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "Findgeometry_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "geometry_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "geometry_msgs"
  with any of the following names:

    geometry_msgsConfig.cmake
    geometry_msgs-config.cmake

  Add the installation prefix of "geometry_msgs" to CMAKE_PREFIX_PATH or set
  "geometry_msgs_DIR" to a directory containing one of the above files.  If
  "geometry_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< cartographer_ros_msgs	[ Exited with code 1 ]
Aborted  <<< turtlebot3_description                    
Aborted  <<< turtlebot3_msgs                           
--- stderr: cartographer                               
diff: /Users/intern/turtlebot3_ws/build/cartographer/AllFiles.cmake: No such file or directory
CMake Error at /usr/local/Cellar/cmake/3.12.2/share/cmake/Modules/FindPkgConfig.cmake:659 (message):
  None of the required 'cairo>=1.12.16' found
Call Stack (most recent call first):
  CMakeLists.txt:36 (PKG_SEARCH_MODULE)

---
Aborted  <<< cartographer
--- stderr: pcl_conversions                                                 
CMake Error at CMakeLists.txt:21 (find_package):
  By not providing "Findsensor_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "sensor_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "sensor_msgs" with
  any of the following names:

    sensor_msgsConfig.cmake
    sensor_msgs-config.cmake

  Add the installation prefix of "sensor_msgs" to CMAKE_PREFIX_PATH or set
  "sensor_msgs_DIR" to a directory containing one of the above files.  If
  "sensor_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Aborted  <<< pcl_conversions

Summary: 1 package finished [1min 21s]
  1 package failed: cartographer_ros_msgs
  4 packages aborted: cartographer pcl_conversions turtlebot3_description turtlebot3_msgs
  3 packages had stderr output: cartographer cartographer_ros_msgs pcl_conversions
  4 packages not processed
The file /Users/intern/turtlebot3_ws/None does not exist.
```
  
8. Please, describe detailedly what difficulty you are in 

I'm able to run the ROS2 talker/listener demo, but I'm unable to build the turtlebot3 installation. As per the logs, it's not able to find the `CMAKE_MODULE_PATH` for `sensor_msgs`, as well as `geometry_msgs`. I'm unsure how to fix these issues to be able to run ROS2 on my Macbook.

nwesthoff, 2018-09-11:
Hi :) thanks for the response.

Running the first command resulted in a `tslf not found` as well as `rttest not found` errors. I suspected these were actually build issues in the ROS2 installation, so I erased the build folders (from ros2_ws as well as turtlebot3_ws), and recreated the ROS2 installation.

The turtlebot3_ws build is now stopping at `cairo not found`, even though it does show up in `/opt/local/include/cairo`. So I'm still unsure on how to proceed unfortunately.

Let me know if I have to create a new issue.

gpwhd07, 2018-11-15:
This issue will be closed, but you can reopen a new issue about your another question about ROS2. Thanks.

closed 2018-11-15 by gpwhd07