ROBOTIS-GIT/turtlebot3 #307: TurtleBot3 Lidar SLAM no map received

timdori, 2018-09-10:
Hey Guys,
i hope you can help me. I have an OpenCr Board and kind of a Mecanum turtlebot3 with an Lidar sensor. I am trying to map my office but its not working. I will try to give u as information as possible.

my ros commands are:
roscore
roslaunch rplidar_ros view_rplidar.launch
export TURTLEBOT3_MODEL=burger
roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping

with rostopic list i get:
/clicked_point
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/move_base_simple/goal
/rosout
/rosout_agg
/scan
/tf
/tf_static
/turtlebot3_slam_gmapping/entropy

rostopic echo /map gets no data

in the gmapping command window the following warning pups up:
[ WARN] [1536589138.659010571]: MessageFilter [target=odom ]: Dropped 100.00% of messages so far. Please turn the [ros.gmapping.message_notifier] rosconsole logger to DEBUG for more information.

when i take fixed_frame laser:
[image]

when i take fixed_frame odom:
[image]

I hope anyone can help me. Thank you.

kijongGil, 2018-09-11:
Hi @timdori :)
If you want to SLAM using Mecanum turtlebot3 with rplidar, you have to solve some problem. First, bringup your turtlebot3. we have release turtlebot3_bringup package. but It is based on LDS and origin turtlebot3. So, you have to modify this code for rplidar. 
Second, you need odom data to do SLAM. But, calculating odom of origin and Mecanum is different. So you have to make a new calculating odom. refer to this code https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino

Thanks,
Gilbert

timdori, 2018-09-11:
Hi Gilbert

Thank you for the fast reply.

1.) I tried:
roslaunch turtlebot3_bringup turtlebot3_robot.launch
but it doenst change anything. Is it possible that my general rviz settings are wrong?
I have to choose fixed frame odom?

2.) Im using a mecanum turtlebot version from the ipa. They implemented the odom data. So my Rob is already publishing odom data. Is it possible to check whether odom has to be transformed with tf? or to check whether it is in a correct datatype?

Thank you,
Tim

kijongGil, 2018-09-11:
I think that mecanum turtlebot doesn't matter, if you use code you said.
Unfortunately I didn't use rplidar. But, I think you have to setup rplidar frame_id = `base_scan` and Fixed frame in Global Obtions =`map`.

timdori, 2018-09-12:
I have got a last little problem... i tested the mapping with my remote pc and it worked perfectly. But when i try to scan with the same lidar connected to my BotPc i got strange rotational shift effects. I think there has to be some kind of tf transformation which was not been there before.

REMOTE PC directly connected to Lidar:
![45303928-3bfb4b80-b517-11e8-9a38-47fc333e2ecb](https://user-images.githubusercontent.com/42413678/45421458-6111ca80-b68d-11e8-94ac-44d2f2ec5f9c.png

BOT PC directly connected to Lidar:
- there are often laserpoints in radial kind far away
- when i try to map the botmodel change direction each time the new scan data comes in

[image]

Any ideas?

timdori, 2018-09-12:
also SOLVED there was an node in the upstart that conflicted this situation :) Thank you very much!

kijongGil, 2018-09-13:
You mean that cloud point keeps spinning when you run the rviz in BotPc? Or just lotated?

timdori, 2018-09-13:
Yes the point cloud was spinning in the same frequence as the pointcloud was refreshing. So it came to a few points in the point cloud which were in radial kind on the maximum range.
And when I tried to map i saw the robotmodel was rotating with the same frequenz.
But when killed all running processes and tried after this all was fine.
So I think it will be all fine when i clean the ros upstart. Tomorrow i will check it.

timdori, 2018-09-14:
Indead there was a node which conflicted the mapping. I killed the roscore and all was fine.
Now I am in another issue. When i try to navigate the robotmodel in rviz is not moving but the real rebot is each time moving on a slightly curve. But because of the wrong situation for the model. There are wrong commands send to the real robot. Any idea how fix this?

kijongGil, 2018-09-17:
If you use tutlebot3_mecanum_core.ino, you need a URDF mecanum ver. Because origin turtlebot3 has two wheels, mecanum has four wheels. So, you have to make two joint state of wheel.
If you only need a navigation without rviz. you only need a odom topic. Because navigation package subscriber only odom topic. refer to follow link.
http://wiki.ros.org/nav_core
And, you have to set parameters for your turtlebot3.

timdori, 2018-09-17:
I have changed my firmware so it is now publishing /odom. But my amcl_pose topic is still only publishing the estimate position. And my odom is publishing each time the same. This both happens even when the robot is moving. So my robot model in rviz is not moving but i get no errors anymore...

kijongGil, 2018-09-19:
Your rviz is get no error? How about cloud point? is it not moving?

timdori, 2018-09-28:
All is running now. The problem was a wrong odom.

robotpilot, 2018-09-30:
Cool! It is good to hear back from you. I am glad your problem has been resolved. I will close this issue. I hope everything is going great. Thanks.

closed 2018-09-30 by robotpilot

harshitkumawat444, 2020-01-23:
> SOLVED IT :)
> After a reboot and frame_id = base_scan and changing the parameters from odom to odom_combined the mapping is working!
> Thank you very much!

how did u changed the frame_id for rplidar and i am using hector slam so do i need to the change the odom parameter 

i am new in this field and i am facing same problem of " no map received'

timdori, 2020-01-23:
There should be a Launch File in which you can change the parameter for the frame_id. With the help of rosnode info for the mapserver you should get the informations of the odom topic you need.

So checkout which topics and which frames are needed for your slam and then edit yours.

Dhaour9x, 2020-05-12:
> 
> 
> There should be a Launch File in which you can change the parameter for the frame_id. With the help of rosnode info for the mapserver you should get the informations of the odom topic you need.
> 
> So checkout which topics and which frames are needed for your slam and then edit yours.

I am facing the same issue, can you pls make same screenshots about the changes, which you have changed. I need /odom for SLAM.
here turtlebot_slam.launch and turtlrbot3_gmapping.launch
[turtlebot3_gmapping.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/4616285/turtlebot3_gmapping.txt)

[turtlebot3_slam.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/4616278/turtlebot3_slam.txt)

here rostopic list
/battery_state
/cmd_vel
/cmd_vel_rc100
/diagnostics
/firmware_version
/imu
/joint_states
/magnetic_field
/motor_power
/odom
/reset
/rosout
/rosout_agg
/rpms
/scan
/sensor_state
/sound
/tf
/version_info
and when I run : roslaunch turtlebot3_slam turtlebot3_slam I got the following results
[image]
[image]

I hope anyone can help me. Thank you.