ROBOTIS-GIT/turtlebot3 #308: CARTOGRAPHER_CMAKE_DIR does not exist error during ROS2 build

nwesthoff, 2018-09-11:
1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: xxx
    - Firmware version: 1.2.0
 
6. Write down the commands you used in order
After trying and failing to install on MacOS ([issue](https://github.com/ROBOTIS-GIT/turtlebot3/issues/306)), I decided to run an Ubuntu 18.04 install, but ran into different problems. 

   1. First went through the full ROS2 on OS X installation [guide](https://github.com/ros2/ros2/wiki/OSX-Development-Setup)
   2. added `source ~/ros2_ws/install/local_setup.bash` to bash profile and sourced it
   3. created `turtlebot3_ws` and cloned git repos
   4. ran `cd ~/turtlebot3_ws && colcon build`

 
7. Copy and Paste your error message on terminal
```bash
Starting >>> cartographer
Finished <<< cartographer [49.9s]                          
Starting >>> cartographer_ros_msgs
Finished <<< cartographer_ros_msgs [17.6s]                              
Starting >>> pcl_conversions
Finished <<< pcl_conversions [8.76s]                              
Starting >>> turtlebot3_description
Finished <<< turtlebot3_description [1.97s]                         
Starting >>> turtlebot3_msgs
Finished <<< turtlebot3_msgs [6.65s]                              
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_teleop [0.69s]                 
Starting >>> cartographer_ros
--- stderr: cartographer_ros                                
CMake Error at /usr/lib/cmake/eigen3/Eigen3Config.cmake:24 (message):
  File or directory /usr/share/cartographer/cmake referenced by variable
  CARTOGRAPHER_CMAKE_DIR does not exist !
Call Stack (most recent call first):
  /home/turtle/turtlebot3_ws/install/cartographer/share/cartographer/cartographer-config.cmake:57 (set_and_check)
  CMakeLists.txt:39 (find_package)

---
Failed   <<< cartographer_ros	[ Exited with code 1 ]

Summary: 6 packages finished [1min 28s]
  1 package failed: cartographer_ros
  1 package had stderr output: cartographer_ros
  3 packages not processed

```
  
8. Please, describe detailedly what difficulty you are in 
Similar issue is described [here](https://github.com/googlecartographer/cartographer_ros/issues/385). I've tried running the executor sequentially (to ensure other processes are finished), but the same problem persists.

routiful, 2018-09-12:
Hi :)

Did you installed `eigen3`?  

```
$ sudo apt install libeigen3-dev
```

nwesthoff, 2018-09-12:
Yes I did, I had to install ceres-solver before, which has `eigen3` as a dependency.

nwesthoff, 2018-09-12:
I saw there were new commits on the turtlebot3 repo, which I pulled (in case that was a fix), however, now I'm also getting another error:

```bash
Starting >>> cartographer
Starting >>> cartographer_ros_msgs
Starting >>> pcl_conversions
Starting >>> turtlebot3_description
Starting >>> turtlebot3_msgs
Starting >>> turtlebot3_lidar
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_teleop [1.17s]                                                                 
[2.189s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'turtlebot3_lidar' because it has no 'install' target
--- stderr: turtlebot3_lidar
CMake Warning at CMakeLists.txt:23 (message):
  Can not compile example: The profiles PROFILE_CREATE_ENTITIES_REF,
  PROFILE_CREATE_ENTITIES_XML, PROFILE_READ_ACCESS, and PROFILE_UDP_TRANSPORT
  must be enabled.

---
Finished <<< turtlebot3_lidar [1.78s]
Finished <<< turtlebot3_description [3.32s]                                                                
Starting >>> turtlebot3_bringup
Finished <<< turtlebot3_bringup [13.0s]                                                                          
Finished <<< turtlebot3_msgs [24.6s]                                                                              
Starting >>> turtlebot3_node
Finished <<< pcl_conversions [28.9s]                                                                              
Finished <<< turtlebot3_node [22.1s]                                                                              
Finished <<< cartographer_ros_msgs [56.9s]                                                                       
Finished <<< cartographer [1min 14s]                                
Starting >>> cartographer_ros
--- stderr: cartographer_ros                                
CMake Error at /usr/local/lib/python3.6/dist-packages/cmake/data/share/cmake-3.12/Modules/FindPkgConfig.cmake:659 (message):
  None of the required 'yaml-cpp>=0.5.1' found
Call Stack (most recent call first):
  CMakeLists.txt:37 (pkg_search_module)

CMake Error at /usr/lib/cmake/eigen3/Eigen3Config.cmake:24 (message):
  File or directory /usr/share/cartographer/cmake referenced by variable
  CARTOGRAPHER_CMAKE_DIR does not exist !
Call Stack (most recent call first):
  /home/turtle/turtlebot3_ws/install/cartographer/share/cartographer/cartographer-config.cmake:57 (set_and_check)
  CMakeLists.txt:39 (find_package)

---
Failed   <<< cartographer_ros	[ Exited with code 1 ]

Summary: 9 packages finished [1min 17s]
  1 package failed: cartographer_ros
  2 packages had stderr output: cartographer_ros turtlebot3_lidar
  1 package not processed
```

I'm not sure whether these have anything to do with eachother, but it might help.

nwesthoff, 2018-09-17:
Okay, so... I've determined it's an issue with the install folder. For me it was installing the cartographer packages in `/usr/local/cartographer/cmake` and library in `/usr/lib/local/` instead of share. For now I symlinked them to the folder where cmake expects them, but it should probably be fixed (though I'm not sure how).

I hope you can make the install process a little more reliable with this information.

routiful, 2018-09-18:
Thank you for sharing your experience :)
I will test it and supplement our emanual
Thanks again :)

closed 2018-09-18 by routiful

alsora, 2018-09-25:
A quick solution I found consists in modifying a file inside Cartographer repo.

 `cartographer/cartographer-config.cmake.in` is the file.

Make sure that the macro `set_and_check(CARTOGRAPHER_CMAKE_DIR "@PACKAGE_CARTOGRAPHER_CMAKE_DIR@")`
is at the beginning of the file, i.e. right after the `@PACKAGE_INIT@` line.

d01010101, 2018-09-28:
So I just followed the tutorial at http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#ros2 and the error is still there:

CMake Error at /usr/lib/cmake/eigen3/Eigen3Config.cmake:24 (message):
  File or directory /usr/share/cartographer/cmake referenced by variable
  CARTOGRAPHER_CMAKE_DIR does not exist !
Call Stack (most recent call first):
  /home/art/turtlebot3_ws/install/cartographer/share/cartographer/cartographer-config.cmake:57 (set_and_check)
  CMakeLists.txt:39 (find_package)

alsora's workaround does not work for me.

Still, this issue is closed... what should I do? Could you tell me it clearly please, I do not know what cmake expects etc., there is a good reason for which I came here from a tutorial... thanks :)

alsora, 2018-09-28:
@d01010101 It's not Eigen that wants something from Cartographer, but Cartographer expects to be built in a standard directory (e.g. `/usr/local/`) while instead it is built in your ROS2 workspace.

You are cloning the correct repository, which actually is not the latest source, but is the one required to be used with the `turtlebot_cartographer` configuration files.

For what concerns my fix, I'm running it in a Docker container with the following command 
`(echo 33 m 22; echo wq) | ex cartographer/cartographer-config.cmake.in`
In practice I'm moving line 33 to become line 22.

mkhansen-intel, 2018-10-30:
@alsora - your fix worked for me! Could you submit that as a patch upstream to the cartographer repo?

alsora, 2018-10-31:
@mkhansen-intel actually I got the idea for the fix from the master branch of the cartographer repo.
This problem is only present for the `release-latest` tag. 

I think that it would be more useful to update the turtlebot stuff to work with the master... It should only be a matter of creating a new lua config file