ROBOTIS-GIT/turtlebot3 #310: turtlebot3_follower does not work, and always return with 'empty'

kfr2tom, 2018-09-12:
Hello, we use Turtlebot3 with **Kinetic on Ubuntu 16.04**.

```
## Predict the position
[x for (x , y) in self.labels.iteritems() if y == self.clf2.predict(laser_data_set) ] 
print 'check_people return x:',x
print 'check_people return y:',y
```
```
[x for (x , y) in self.labels.iteritems() if y == self.clf.predict(data_test_set) ] 
print 'laser_scan return x:',x
return x
```

we have add some print to output the “x” and “y” from the function [check_people(self)](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L39-L47) and [laser_scan(self)](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L67), But all of them are "empty", as following list.

> check_people return x: empty
> check_people return y: 7
> laser_scan return x: empty
> check_people return x: empty
> check_people return y: 7
> laser_scan return x: empty

And we found a strange point, the value of [ret](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L41-L47) in **check_people(self)** is always 1, it means the node always consider check_people successfully. Is it wrong?
  
Can you explain how to use the [clf](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/clf) and [clf2](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/clf2) to achieve [laser_scan(self)](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L49) and [check_people(self)](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L19) correctly? Or how to implement the forecasting process? Thanks for read first!

kijongGil, 2018-09-13:
Hi @kfr2tom :)
I checked it you said. It is wrong,  you're right.
Thanks for your contribution, we will update this solution.

Gilbert.

kfr2tom, 2018-09-14:
> Hi @kfr2tom :)
> I checked it you said. It is wrong, you're right.
> Thanks for your contribution, we will update this solution.
> 
> Gilbert.

Hi @kijongGil 
Thanks for your help, so how can I use the [clf](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/clf) and [clf2](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/clf2) to achieve [laser_scan(self)](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L49) and [check_people(self)](https://github.com/ROBOTIS-GIT/turtlebot3_applications/blob/kinetic-devel/turtlebot3_follower/scripts/follower.py#L19) correctly?

Tom.

kijongGil, 2018-09-17:
This code uses scikit-learn. I think this will be help you.
http://dataaspirant.com/2017/02/13/save-scikit-learn-models-with-python-pickle/

closed 2018-09-30 by robotpilot