ROBOTIS-GIT/turtlebot3 #311: Bigger wheels for friends

pitosalas, 2018-09-12:
Hi! This may not be the right place to post this... But let me try. 
We are modifying our TB3 and we'd like to put bigger wheels on the modified "friend". We  want to have greater clearance.

Does Robotis sell wheels that would fit the four screw setup but that are around 4" in diameter? We know this will affect other things, but for now I am just wondering about the wheels.

kijongGil, 2018-09-13:
Hi @pitosalas :)
Unfortunately, we do not sell any other wheels besides the existing wheels.
Do you know turtlebot3 friends : carrier? http://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#locomotion
I do not know what wheel size you want, but we release stl files of carrier wheel for 3D printing .

Thanks,
Gilbert.

kijongGil, 2018-09-20:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-09-20 by kijongGil

stevensu1838, 2019-01-10:
Hi @kijongGil , I am also trying to put on bigger wheels. May I please ask what part of the code in TB3 burger package shall I modify after changing the wheels?  Thx

bergercookie, 2020-10-09:
Not very relevant to the subject but, I want to use the wheels suggested for the TB3 Carrier setup (https://docs.google.com/spreadsheets/d/1vbIFMRJMbyOd2D6BtwV1Ow_yBc9xRHktqZhTTmwI_l4/edit#gid=71921372). Do you know where can I buy matching tires for these wheels? The dimensions for the carrier wheels are 118mmx18mm and they are 3d-printed