ROBOTIS-GIT/turtlebot3 #312: i use

geerniya, 2018-09-13:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [y ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [y ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [y ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [y ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: ros kinetic
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - export TURTLEBOT3_MODEL=waffle
    - roslaunch turtlebot3_gazebo turtlebot3_world.launch
    - roslaunch turtlebot3_slam turtlebot3_slam.launch
    - rosrun rviz rviz -d `rospack find turtlebot3_slam`/rviz/turtlebot3_slam.rviz
    - roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
   
 
7. Copy and Paste your error message on terminal

    -HERE
  
8. Please, describe detailedly what difficulty you are in 

    -  When I type in the above code, the robot can be displayed, can also operate through the keyboard in gazebo. But the slam map does not display in the rviz. I do not know what went wrong. Can you help me ? Thanks.
[image]
[image]

When I modify the fixed frame to odom , the model displays in rviz and the laser-scan map is correct. But where is the slam map?
[image]

kijongGil, 2018-09-13:
Hi @geerniya :)
First, the command you used is old version. please update our turtlebot3 packages.
If you have a same error after trying again new command(http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#ros-packages-for-gazebo), please show me the topic list
You can check your topic list by this command :
`rostopic list`

Thanks,
Gilbert

kijongGil, 2018-09-17:
Hmm... It's strange. It does not seem to be a problem. If you have a real turtlebot3, try again with real turtlebot3. And did you set master and host IP in ~/.bashrc?

robotpilot, 2018-09-30:
@geerniya, I am glad that the problem has been solved. I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue. Thanks.

closed 2018-09-30 by robotpilot