ROBOTIS-GIT/turtlebot3 #313: Realsense 2 SDK support for Turtlebot3

ShettyHarapanahalli, 2018-09-17:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [*] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [*] Raspberry Pi 3
    - [*] Ubuntu 16.04 

3. Which OS you installed in SBC?

    - [*] Ubuntu MATE 16.04 or later
    - [*] Raspbian
    - [*] Ubuntu MATE 16.04

4. Which OS you installed in Remote PC?

    - [* ] Ubuntu 16.04 LTS (Xenial Xerus)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

roscore   
roslaunch realsense2_camera rs_camera.launch 
  [cntl + c to terminate the propgram as there was not error and launch it with Turtle3  ]
roslaunch turtlebot3_bringup turtlebot3_realsense.launch
 
7. Copy and Paste your error message on terminal
I have the RS 
roslaunch turtlebot3_bringup turtlebot3_realsense.launch
... logging to /home/sshetty/.ros/log/7b2dd07a-ba7b-11e8-a346-54bf64471cc6/roslaunch-sshetty-DL-26493.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sshetty-DL:37001/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/nodelet_manager-1]: started with pid [26511]
process[camera/driver-2]: started with pid [26512]
[ INFO] [1537194029.197471969]: Initializing nodelet with 12 worker threads.
[ERROR] [1537194029.451156944]: /camera/driver - No cameras detected!
[ INFO] [1537194029.451555765]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194034.671065814]: /camera/driver - No cameras detected!
[ INFO] [1537194034.671462255]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194039.891030559]: /camera/driver - No cameras detected!
[ INFO] [1537194039.891393799]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194045.111038060]: /camera/driver - No cameras detected!
[ INFO] [1537194045.111402563]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194050.331044453]: /camera/driver - No cameras detected!
[ INFO] [1537194050.331411322]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194055.551040558]: /camera/driver - No cameras detected!
[ INFO] [1537194055.551432593]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194060.770989573]: /camera/driver - No cameras detected!
[ INFO] [1537194060.771341045]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194065.990992525]: /camera/driver - No cameras detected!
[ INFO] [1537194065.991358522]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194071.211063815]: /camera/driver - No cameras detected!
[ INFO] [1537194071.211472697]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194076.431075895]: /camera/driver - No cameras detected!
[ INFO] [1537194076.431411475]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194081.651008425]: /camera/driver - No cameras detected!
[ INFO] [1537194081.651338315]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194086.871063821]: /camera/driver - No cameras detected!
[ INFO] [1537194086.871471786]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194092.090952896]: /camera/driver - No cameras detected!
[ INFO] [1537194092.091232046]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194097.311045494]: /camera/driver - No cameras detected!
[ INFO] [1537194097.311440523]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194102.531092925]: /camera/driver - No cameras detected!
[ INFO] [1537194102.531481775]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194107.751153471]: /camera/driver - No cameras detected!
[ INFO] [1537194107.751560675]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537194112.971028298]: /camera/driver - No cameras detected!
[ INFO] [1537194112.971387836]: /camera/driver - Sleeping 5 seconds then retrying to connect
[roslaunch-sshetty-DL-26493.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/2388929/roslaunch-sshetty-DL-26493.log)
[roslaunch-sshetty-DL-26493.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/2388930/roslaunch-sshetty-DL-26493.log)
[roslaunch-sshetty-DL-26493.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/2388934/roslaunch-sshetty-DL-26493.log)

  
8. Please, describe detailedly what difficulty you are in 

    - Does turtle 3 support realsense 2 sdk, we have new SDK from real sense [https://github.com/IntelRealSense/librealsense]

kijongGil, 2018-09-19:
Hi @ShettyHarapanahalli :)
We will support realsense 2 SDK in the near future.
I'm sorry for the inconvenience.

Gilbert.

kijongGil, 2018-09-20:
I'm going to close this issue. I will reopen it when we support realsense 2 sdk.

closed 2018-09-20 by kijongGil