ROBOTIS-GIT/turtlebot3 #315: Turtle 3 without open cr board using only PI and my own motor driver

ShettyHarapanahalli, 2018-09-18:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [*] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [*] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [*] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [*] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    -Can I build my Turtle 3 without open cr board using only PI and my own motor driver.

kijongGil, 2018-09-19:
Hi @ShettyHarapanahalli :)
We released all information about OpenCR.
https://github.com/ROBOTIS-GIT/OpenCR
http://emanual.robotis.com/docs/en/parts/controller/opencr10/

But, I recommend that using OpenCR. Because It will take a lot of time and effort to develop a new motor driver.

Thanks,
Gilbert

ShettyHarapanahalli, 2018-09-24:
Ok thanks for the response.

closed 2018-09-24 by ShettyHarapanahalli