ROBOTIS-GIT/turtlebot3 #316: TF2 transform issue with turtlebot3_cartographer.launch

sleepydbear, 2018-09-19:
I'm trying to run google cartographer with Turtlebot3 simulation on gazebo (ros melodic and ubuntu 18). Coming across this warning on the terminal when i run turtlebot3_cartographer.launch:

`Warning: Invalid argument "/map" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 134 in /tmp/binarydeb/ros-melodic-tf2-0.6.3/src/buffer_core.cpp`

Also, the map topic does not publish anything. Any idea why this is happening? I gather from [cartographer documentation](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) that only TF2 is supported with ROS 

Thanks in advance!

routiful, 2018-09-27:
Hello :) 

Sorry for late reply.
I didn't test your development environments but I assume you are supposed to modify below launch file related topic name from `/map` to `map`

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/global_costmap_params.yaml

sleepydbear, 2018-10-01:
Solved the problem! thanks 
However there seems to be a lot of other TF problems with cartographer library, and the instructions on the TB3 manual for using cartographer doesn't seem to be adequate. Will try to get it running. Cheers :)
-Sai

closed 2018-10-01 by sleepydbear

routiful, 2018-10-02:
Great! If you have any further question, please open new issue :)

routiful, 2018-10-15:
@ZhangLiangyu5411 you don't need to modifiy that in Kinetic distro.

ZhangLiangyu5411, 2018-10-16:
> @ZhangLiangyu5411 you don't need to modifiy that in Kinetic distro.

Thanks for your reply,and now it works well .thank you ~

dfcknj, 2019-02-28:
Hello! 
I have got an other problem when running google cartographer with Turtlebot3 simulation on gazebo (ros melodic and ubuntu 18)
when I'm launching cartographer with the command:
`roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer configuration_basename:=turtlebot3_lds_2d_gazebo.lua`

`[ WARN] [1551322996.038315946, 4964.699000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.`

Any idea why this is happening?