ROBOTIS-GIT/turtlebot3 #319: Need Help wont run bringup, cannot diagnose myself.

haldaneduncan, 2018-09-22:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
   
2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
 

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial
  

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
  Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial

5. Write down software version and firmware version

    - Software version: 16.04.5 on both 
    - Firmware version: [x.x.x] i didn't figure out how to do this 
 
6. Write down the commands you used in order

    roslaunch turtlebot3_bringup turtlebot3_robot.launch

ROSlaunch runs 

then i go to run Rviz using 
$ export TURTLEBOT3_MODEL=${TB3_MODEL}
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz

 
7. Copy and Paste your error message on terminal

 while processing /home/pi/catkin_ws/src/turtlebot3/turtlebot3_bringup/launch/includes/description.launch.xml:
Invalid <arg> tag: turtlebot3_description
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share. 

Arg xml is <arg default="$(find xacro)/xacro --inorder '$(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro'" name="urdf_file"/>
The traceback for the exception was written to the log file

  
8. Please, describe detailedly what difficulty you are in 

    I'm trying to run the turtlebot for the first time, getting the above message when running the bringup portion from instructions, I've checked and rechecked the Ip address and reinstalled or started from scratch multiple time seeing if it was something i have done or neglected, Which i suspect i just cannot find. 

http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#run-roscore

haldaneduncan, 2018-09-22:
I just found that i'm using Arduino 1.8.7 on the Remote PC, if this makes a difference?

routiful, 2018-09-27:
Hi :)

You are supposed to type `TB3_MODEL`. For example

```
$ export TURTLEBOT3_MODEL=burger
```

or

```
$ export TURTLEBOT3_MODEL=waffle_pi
```

haldaneduncan, 2018-09-29:
yes, i believe i was following the instructions correctly and  what you typed is what i used, and its a burger
.

routiful, 2018-09-30:
http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#load-a-turtlebot3-on-rviz

Above instructions are supposed to run your RemotePC not Raspberry Pi.
I guess you ran that in Raspberry Pi

```
/home/pi/catkin_ws/src/turtlebot3/turtlebot3_bringup/launch/includes/description.launch.xml:
```

haldaneduncan, 2018-09-30:
my remote PC is another second Raspberry pi running ubuntu, i did not want to run a dual boot on my PC, or try to run my laptop with this as so far as i could see at the time i started this it wouldn't work. Do you suggest doing this? are there instructions on the emanual for a achieving a windows install that can run the roscore and do the necessary functions?

routiful, 2018-10-02:
Sorry, we highly recommended use another PC to run other packages specially Rviz. Because Raspberry Pi 3 is not enough to run visualization software.

haldaneduncan, 2018-10-04:
routiful, 

Thank you for your  help, used a desktop and I can at least get bringup to work. working on slam and other functions , the basic functions work perfectly on the ones i tried, the driving one was a great satisfaction after struggling with this.