ROBOTIS-GIT/turtlebot3 #320: hls_lfcd_lds_driver/hlds_laser_publisher error on roslaunch turtlebot3_bringup turtlebot3_robot.launch

BuzzTechnology, 2018-09-23:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [X] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

roslaunch turtlebot3_bringup turtlebot3_robot.launch
 
7. Copy and Paste your error message on terminal

Please see below.
  
8. Please, describe detailedly what difficulty you are in 

I cannot launch ROS on my TurtleBOT.

--------------------

When first trying to run 'roslaunch turtlebot3_bringup turtlebot3_robot.launch' on my Raspberry Pi/Ubuntu Mate turtlebot3 as per the eManual, I got the two errors below:

ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver

and

[turtlebot3_core-1] process has finished cleanly

The terminal full output is shown below.

Could this is related to an initial failure when running the 'rosrun turtlebot3_bringup create_udev_rules' command that allows the openCR board to access the Raspberry Pi port without root permission? When I first ran it without the OpenCR board connected and before the openCR setup, it failed. I then ran it afterwards with everything connected and it seemed to work fine...

================================================

TERMINAL OUTPUT
sa@sa-ThinkPad-X230:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch ... logging to /home/sa/.ros/log/df6065d4-b5db-11e8-87be-606720897a0c/roslaunch-sa-ThinkPad-X230-2980.log Checking log directory for disk usage. This may take awhile. Press Ctrl-C to interrupt Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.13:43757/

SUMMARY
PARAMETERS * /rosdistro: kinetic * /rosversion: 1.12.14 * /turtlebot3_core/baud: 115200 * /turtlebot3_core/port: /dev/ttyACM0 * /turtlebot3_core/tf_prefix: * /turtlebot3_lds/frame_id: base_scan * /turtlebot3_lds/port: /dev/ttyUSB0

NODES / turtlebot3_core (rosserial_python/serial_node.py) turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics) turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.13:11311

process[turtlebot3_core-1]: started with pid [2997] ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver ROS path [0]=/opt/ros/kinetic/share/ros ROS path [1]=/home/sa/catkin_ws/src ROS path [2]=/opt/ros/kinetic/share process[turtlebot3_diagnostics-3]: started with pid [2998] [turtlebot3_core-1] process has finished cleanly log file: /home/sa/.ros/log/df6065d4-b5db-11e8-87be-606720897a0c/turtlebot3_core-1*.log

routiful, 2018-09-28:
Hi :)

Of course, you are supposed to run `create_udev_rules` node in order to usb permission.
Did you resolve your problem after run that?

senake, 2018-09-30:
Hi there,

Thank you. That did resolve the problem.

I am now, unfortunately, stuck on another issue when running 'roslaunch turtlebot3_example turtlebot3_pointop_key.launch'- as below - but will raise a separate issue for it.

[INFO] [1538312797.554481]: Cannot find transform between odom and base_link or base_footprint
[INFO] [1538312799.178262]: shutdown program.
[turtlebot3_pointop_keyboard-1] process has finished cleanly
log file: /home/sa/.ros/log/504de2c6-c4af-11e8-a8d9-606720897a0c/turtlebot3_pointop_keyboard-1*.log

routiful, 2018-10-02:
Great :) If you have any further question, please open issue.

closed 2018-10-02 by routiful