ROBOTIS-GIT/turtlebot3 #321: TurtleBot3 eManual doesn't include set up instructions for the UP SBC in the Realsense kit

OriginalSen, 2018-09-23:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [X] Waffle - my purchase from Robosavvy was a Waffle Pi kit with Realsense kit to upgrade the SBC and add the R200 depth camera. 
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [X] Intel Atom based UP board (included in the Intel Realsense R200 developer kit provided by Robosavvy with the TurtleBot3) 

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [X] See below - I am not sure - advice needed as the Robotis eManual only has instructions for the Intel Joule. 

4. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    See (8) below. Instructions for Up SBC install are needed - software  & firmware is not yet installed.

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    Instructions for Up SBC install needed - software  & firmware not yet installed.
 
7. Copy and Paste your error message on terminal

    Not applicable.
  
8. Please, describe detailedly what difficulty you are in 

There are no instructions in the Robotis eManual for the set up of the Up board that was supplied with the Realsense kit that I was sent with my order. This page for the set up of the single board computer from the manual - http://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/ only has instructions are for the Intel Joule 570X. The steps for the Intel Joule (http://emanual.robotis.com/docs/en/platform/turtlebot3/joule_setup/#joule-setup) refers to this link (https://software.intel.com/en-us/flashing-the-bios-on-joule) for a BIOS update in which the board is flashed by connection to a PC via the USB-C connection. 

The only thing that I could find online relating to both the Up and Turtlebot3 was this post - https://community.robotsource.org/t/convert-a-turtlebot3-waffle-pi-to-use-up-board-with-r200-camera/1397/4 - where one of the Robotis team, Leon_Ryuwoon_Jung seems to have said that they tried this combination back in 2016. 

Please can you point  me to an updated version of the eManual where they detail the Up board set up? Please also let me know which connectors (USB3 on-the-go and other) to use for both the BIOS upgrade (if needed) and Turtlebot3 build.

PS. I found the procedure for flashing the BIOS on the UP that I found on their Wiki web site (https://wiki.up-community.org/Bios_Update) seem to recommend that it is done without a connection straight off a USB drive. Is it relevant?

routiful, 2018-09-30:
Hi :)

We haven't provided instructions about Up-board setup. But we have tested it worked fine.(RobotSource user have done too before)

However, we will upload `How to set up in Up board` in e-manual after two weeks later because we are going to attend IROS.

Sorry for inconvenience.

kijongGil, 2018-10-05:
I will close this issue once. Please reopen or register a new issue at any time.

closed 2018-10-05 by kijongGil