ROBOTIS-GIT/turtlebot3 #324: No Waffle Pi OpenCR firmware

d01010101, 2018-09-28:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [] Raspbian
    - [x] onw from http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#ros2

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19

5. Write down software version and firmware version

    - Software version: no idea
    - Firmware version: no idea
 
6. Write down the commands you used in order

    - followed http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#ros2
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

I want to have ROS2 on Waffle Pi. Followed the tutorial, OpenCR has only Burger images. The tutorial says nothing about this.

Anyway, I installed the Burger images and now the robot does not even work with the pilot. No idea why. If Waffle is not yet supported, why is not written clearly? Are you going to update the manuals with instructions also for Waffle? Do you know a date when it is more or less tested? For now I stumped on a "closed" bug https://github.com/ROBOTIS-GIT/turtlebot3/issues/308

nwesthoff, 2018-09-29:
Duplicate of https://github.com/ROBOTIS-GIT/turtlebot3/issues/314

robotpilot, 2018-09-30:
Hi @d01010101, @nwesthoff,
Yes. You are right. We have not yet released the ROS2 support source code for TurtleBot3 Waffle Pi. Our team is currently participating in ROSCon2018. When we return, we will release the source code for TurtleBot3 Waffle Pi. It will take about two weeks because there is also IROS2018 after ROSCon2018.

d01010101, 2018-10-30:
Hi, just a small question, do you have a rough estimate of when the firmware for Waffle will possibly be available? I am not in a hurry, just to arrange the things I am doing now. Thanks!

routiful, 2018-10-31:
Not yet, we are working to make compatible with Micro-XRCE-DDS version 1.0.0.
I assume that the firmware will be announce after ROS2 Crystal.

OpusK, 2018-12-14:
Hi, @d01010101, @nwesthoff.

We release [ros2arduino 0.0.1](https://github.com/ROBOTIS-GIT/ros2arduino/releases/tag/0.0.1) yesterday.
And I have [requested to register with Arduino library manager](https://github.com/arduino/Arduino/issues/8295).

[The waffle firmware](https://github.com/ROBOTIS-GIT/OpenCR/tree/develop/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_waffle/turtlebot3_core) is currently uploaded to the develop branch of OpenCR.
We plan to release it officially in the next week.

The ROS2 node related to TB3 is still in development, and we are also making RaspberryPi image.
But if you want, you can now use the ros2arduino and OpenCR firmware to get the TB3 topic, and you can test it partly through the TB3 node currently under development in the ros2 branch.

closed 2018-12-20 by OpusK