ROBOTIS-GIT/turtlebot3 #325: Navigation-2D Nav Goal doesn't work

hansenmaster, 2018-10-02:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ 1] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [1 ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [1 ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ 1] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version
https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888
    - Software version: [1.0.0]
    - Firmware version: [1.2.0]
 
6. Write down the commands you used in order

[PC] roscore

[Turtlebot3] roslaunch turtlebot3_bringup turtlebot3_robot.launch

make mapping SLAM
[PC]
->roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping

then use Android teleop and get this map
[image]
closed gmapping node

run navigation in rviz
->roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml

->Set inital pose and move robot with teleoperation, until it seems the lidar scan line up well with the map
-> kill the teleop node and set the 2d nav goal

My final rviz pic
[image]

7. Copy and Paste your error message on terminal

    - in RViz Costmap : warning map not found
    - Costma2DROS trasnfor timeout , over and over again
  
8. Please, describe detailedly what difficulty you are in 

    - After initial pose setting i set 2d nav goal the robot doesn't move

My questions:

A. How do i know my initial set pose is good to go? Does it say anyhing when the initial pose is good? Like color or text? Does AMCL localization working?

B. Is the warning in RViz cost map ignorable?

C. I already killed the teleop to make the cmd_vel only come from the move_base function, the /move_base_simple/goal does change when i set the 2d nav_goal. but still the robot doesn't move.
[image]
The robot should just move to the 2d nav goal(red arrow) right away right?

D. I make the LIDAR in higher position, is it okay to start the navigation without change the URDF/tf first? I think it's only 2d so it should be okay?

 I know DoNck has asked the same topic here https://github.com/ROBOTIS-GIT/turtlebot3/issues/266 , but  still i get error even after I tried all the suggestion.
Please guide me by answering my questions, thanks

routiful, 2018-10-02:
Hi :)

A.  You can set initial pose of robot in AMCL launch file. 
https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_navigation/launch/amcl.launch#L4-L6
I recommend that the robot put the position when you start SLAM. 

B. I don't know which warning you said. Please show me the messages

C. I assume that the reason of failed localization makes this problem. If it succeed, the robot move to goal pose right after you set 2d_nav_goal

D. You can modify the origin of base_scan. But this is not critical.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro#L142

hansenmaster, 2018-10-03:
> Hi :)
> 
> A. You can set initial pose of robot in AMCL launch file.
> 
> [turtlebot3/turtlebot3_navigation/launch/amcl.launch](https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_navigation/launch/amcl.launch#L4-L6)
> 
> Lines 4 to 6 in [467c76b](/ROBOTIS-GIT/turtlebot3/commit/467c76bc4fa2e34162f57107388839d82d3bcc0e)
>  <arg name="initial_pose_x" default="0.0"/> 
>  <arg name="initial_pose_y" default="0.0"/> 
>  <arg name="initial_pose_a" default="0.0"/> 
> 
> I recommend that the robot put the position when you start SLAM.
> 
> B. I don't know which warning you said. Please show me the messages
> 
> C. I assume that the reason of failed localization makes this problem. If it succeed, the robot move to goal pose right after you set 2d_nav_goal
> 
> D. You can modify the origin of base_scan. But this is not critical.
> 
> [turtlebot3/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro](https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro#L142)
> 
> Line 142 in [467c76b](/ROBOTIS-GIT/turtlebot3/commit/467c76bc4fa2e34162f57107388839d82d3bcc0e)
>  <origin xyz="0 0 0.0" rpy="0 0 0"/>

Hi routiful :D , i appreciate your fast response

Here i will response your answers :
A. I tried to set initial pose the robot around the start SLAM, it worked great (the particle cloud really dense and close to the actual robot position
[image]

B. Sorry forget to mention the warning, these are the warning
[image]
[image]
Is this ignorable?

C. Now the particle cloud estimation real close to the actual robot, i think the localization worked great. I tried the rostopic pub to send value to /cmd_vel it moved correctly. But when i set the 2d nav goal in RViz, even i see the topic move_base_simple/goal is updated, but i think this don't go to /cmd_vel.
Have any idea to find the possible cause?

D. I'll try to change the origin of base_scan, yeah i think because the map is 2d, this is not critical

hansenmaster, 2018-10-03:
Hi routiful, unfortunately i can't open the link you sent :(

routiful, 2018-10-04:
I recommend connect Internet and ntp. But you can try [chrony](https://chrony.tuxfamily.org/doc/3.3/chrony.conf.html) also.

hansenmaster, 2018-10-04:
Hi routiful.

I've solved the problem with time sync :D . Thank you thank you very much for helping me.
I set both my pc and turtlebot by timedatectl to set both in Asia/Bangkok and turn on the ntp update :dagger: 
How to close this thread and said it's solved?

closed 2018-10-04 by hansenmaster

routiful, 2018-10-05:
Great. If you have any further problem, please open issue :)

echefulouis, 2022-12-08:
HI @hansenmaster and @routiful  I am having similar issues with my turtlebot as its not moving when i set the 2D Nav Goal. I got the same warning about costmap. The solution to yours was to sync the time right?.