ROBOTIS-GIT/turtlebot3 #326: /battery_state/current does not update

jennuine, 2018-10-05:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?
(We have two Waffle Pi's, one with MATE and the other with Raspbian)
    - [x] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.1.0]
    - Firmware version: [1.1.3]
 
6. Write down the commands you used in order

    - [Remote PC] roscore
    - [Turtlebot] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - [Remote PC] rostopic echo /battery_state
 
7. Copy and Paste your error message on terminal

    - No error
  
8. Please, describe detailedly what difficulty you are in 

We have two Waffle Pi's, one with MATE and the other with Rasbian. Both have the same software and firmware versions. We need updates to current from the /battery_state topic but it is always displayed as zero. The only subtopics that update from /battery_state are the header, voltage, and percentage. All the other subtopics never update.

jennuine, 2018-10-08:
Hi @kijongGil 
I've updated the firmware using the shell script and still no change.

jennuine, 2018-10-15:
Hi @kijongGil 

Yes, so far the turtlebot3 works well with the teleop package. The only issue so far is that the current from the /battery_state is not being outputted and we need that for our research.

kijongGil, 2018-10-17:
I'm sorry I misunderstood your question. [/battery_state](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/BatteryState.html) has voltage, charge, capacity, design_capacity, percentage and more. You want to check it, right?
This message format is standard for battery_state. Unfortunately, we do not use this data, so the data value is zero. If you want to check current, you have to add current sensor and make to publisher.

closed 2018-10-26 by kijongGil