ROBOTIS-GIT/turtlebot3 #328: Turtlebot bringup orange flashing led

alsora, 2018-10-08:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.1.0]
    - Firmware version: [1.2.3]

Hi, I'm trying to run the teleoperation node to move a turtlebot waffle pi.
It was working until few days ago, but now it doesn't.

The problem is that when I run 
`roslaunch turtlebot3_bringup turtlebot3_robot.launch`
 
The USER1 led light starts flashing orange. This was something that was not happening before.

I tried both with the teleoperation node, as well as publishing `geometry_msgs/Twist` on the `cmd_vel` topic.

I was thinking at a hardware connection problem, but if I flash the ROS2 firmware for OpenCR and publish the velocity commands using the ROS2 node, everything works fine.

Do you have any idea why ?
Thanks

routiful, 2018-10-11:
Hello :)

The USER1 led shows state of data transfer between SBC and OpenCR.
This is toggled when you launch turtlebot3_robot.launch file.
So your case is not problem.

I recommend that you can check Dynamixel ID using `turtlebot3_setup_motor`

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3

alsora, 2018-10-11:
Hi, thanks for the reply!
I followed the instructions and I was able to run the tests (the motors were moving).

I noticed a strange behavior: apparently ANY ros node on the turtlebot doesn't receive messages.

If I run on my pc
```
rostopic pub -r 10 /cmd_vel geometry_msgs/Twist  '{linear:  {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}'
```

Then I run on the Raspberry
```
rostopic echo /cmd_vel
```

No messages are received!

Note that if I run the opposite (publisher on raspberry, listener on pc) everything works, even if the ros master remains always on my pc.

Do you have any other idea ?

routiful, 2018-10-15:
I recommend that I attached check list.

1. Network configuration
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#5-network-configuration

2. Test by RC100 controller
http://emanual.robotis.com/docs/en/platform/turtlebot3/teleoperation/#rc100

3. Test using Rqt
http://wiki.ros.org/rqt

Cameronthepiper, 2024-01-08:
Hey did this ever actually get resolved?

I'm having the exact same issue and it was working perfectly yesterday.

Any help would be appreciated : )