ROBOTIS-GIT/turtlebot3 #329: Turtlebot 3 navigation detect fake obstacle and miss obstacle

hansenmaster, 2018-10-09:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - navigation tutorial in Turtlebot3 guide
http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#navigation
 
7. Copy and Paste your error message on terminal

    - no error
  
8. Please, describe detailedly what difficulty you are in 

 I succeeded to do some 2d nav goal (especially when the free path is so wide and no obstacle), and the robot moved precisely to the goal pose.

Problem 1)
However, when the robot moved along narrow path like on this picture, the lidar seems like detect an obstacle ahead even it's a free narrow path, i want to move through this narrow alley but the robot will not move there because it think it's a obstacle. This is not always happening, but more frequently than normal (no false obstacle detected).

Rviz view (detect obstacle ahead)
[image]

Real view (no obstacle ahead)
[image]

Question for problem 1:

Why the turtlebot thinks it encounter obstacle when it's clearly free way ahead?
-Is it because the tolerance of costmap (even it's free way, if it is narrow it cannot be passed occasionaly)? -Is my Lidar not reliable anymore?

Problem 2)
I set my Lidar in 35 cm height, so the laser scan only detect at that height, sometimes i have short obstacle like 20 cm box in the real environment. Because of this, the robot sometimes make trajectory that collide with this short box.

Question for problem 2:  
1. Can I add  like ultrasonic sensor on 20 cm height as complementary distance sensor (this sensor only works for obstacle avoidance in navigation, to alter the trajectory avoiding this short obstacle that missed by laser scan)
2. If yes, is there any tutorial for this situation?

Thank you very much for paying attention to my problem

routiful, 2018-10-11:
Hi :)

You can add distance sensor. I attached related references.

http://emanual.robotis.com/docs/en/platform/turtlebot3/additional_sensors/#ir

hansenmaster, 2018-10-12:
Thank routiful, 

after i attached an additional sensor (proximity sensor), 
can i add this sensor and combine it with laser scan for sensor transform? How to do this?
For example i want the proximity sensor added works as obstacle avoidance parameter for short obstacle avoidance that can't be detected by laser.

[image]

routiful, 2018-10-15:
You can use your sensor by using **sensor topics**. 
We don't have any examples how to combine that. But I think you can simply use /odom topic to avoid obstacle.

hansenmaster, 2018-10-17:
Yeah i think need more research and practice to combine more than one sensor for the scanning.

Could you explain more for use /odom topic for avoid obstacle?

routiful, 2018-10-17:
I think you can get a more information about your issue in [ROS Answers](https://answers.ros.org/questions/)

closed 2018-10-30 by gpwhd07