ROBOTIS-GIT/turtlebot3 #330: Inappropriate Navigation Parameters

yoshimalucky, 2018-10-10:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x] etc (Jetson TX2)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [x] etc (Ubuntu 16.04 Jetpack 3.1)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - roslaunch turtlebot3_navigation turtlebot3_navigation.launch
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - We are in difficulty of TB3 navigation. TB3 goes backward frequently when it perform path planning. Also, it rotates infinitely around the goal.

We solved our issues with these parameter changes:

min_x_vel :0
> prevent backward locomotion
https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/dwa_local_planner_params_waffle_pi.yaml#L5

min_rot_vel: 0.4 
> default value 0.9 is too fast. 0.4 is from TB2.
https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/dwa_local_planner_params_waffle_pi.yaml#L23

We are wondering how you fine tune these parameters set as the default. Any ideas?

routiful, 2018-10-11:
Hi :)

We offer some references for parameter tuning. I attached below

http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#tuning-guide

yoshimalucky, 2018-10-15:
We know about the tutorial and each meaning of the parameters. 

This is the question from our client contacting with us the Japan branch. 

There question was same with these two:
https://answers.ros.org/question/272462/how-to-prevent-navigate-backwards-and-only-rotate-and-move-forwards-for-navigation/
http://ros-users.122217.n3.nabble.com/Navigation-Stack-the-robot-rotates-most-of-the-time-after-receiving-a-goal-td1471418.html

According to the past asset here:
https://github.com/turtlebot/turtlebot_apps/blob/indigo/turtlebot_navigation/param/dwa_local_planner_params.yaml

We set the two parameters shown below:
min_vel_x : 0.0
min_rot_vel : 0.4

PR2 also set min_vel_x equal to 0.0 to prevent backward locomotion in its navigation. Again, our question is how you set the parameters of the local planner. We have already known the meaning of the parameters. We would like to know your criteria for how to set the parameters.

I think this change for the default parameters improve a behavior of TB3 navigation.

routiful, 2018-10-15:
As you can see, TurtleBot3 has lidar has full range(360 degree). 
If the lidar only covered half range like first question you attached or PR2, I was supposed to set zero to `min_vel_x` parameters.

Moreover, TurtleBot3 has round shape. It makes no difference between front and rear. I thought that backward motion is not different forward motion in the case of TB3. Furthermore, It makes more useful than only has forward motion. This is why I set not zero number to `min_vel_x` parameters. 

In the case of second issue you attached, the user makes set lower number to `min_rot_vel` or adjust goal tolerance parameter (xy and yaw). I think this is case by case.

The PR2 and turtlebot is great platform and nice tuning guide but this is not an answer. 
The Parameters may vary depending on platform and environments.

I wish my reply is proper answer what you want to know.
However, if you show me the parameter I set is makes critical problems during navigation and the modification which you propose is a simple answer by that, I will accept your requirement.

yoshimalucky, 2018-10-16:
Thanks for your answer. Your answer is perfect!

Even PR2 and TB2 set min_vel_x equal to 0.0, I understood the reason why you set min_vel_x not equal to zero for TB3. 

However, its locomotion is smoother when it goes from its wheel base side because its ball casters are not durable enough. Therefore, I recommend the min_vel_x setting. For the implementation, it sounds good to add an argument such as "move_forward_only" for turtlebot3_navigation.launch.

For the second issue, I thought we should follow the TB2 parameter because the waffle platform size is almost same with the one of TB2 kobuki base. In most case, this prevents its shaky oscillation around a goal point. But, as you say, it depends on each case, so it does not have an perfect answer.

routiful, 2018-10-17:
> However, its locomotion is smoother when it goes from its wheel base side because its ball casters are not durable enough. 

I agree your opinion. 

> For the implementation, it sounds good to add an argument such as "move_forward_only" for turtlebot3_navigation.launch.

If you request PR, I will review it :)

yoshimalucky, 2018-10-26:
Thanks for your comment. I'm closing this issue after PR.

yoshimalucky, 2018-11-05:
Thanks for merging PR!

closed 2018-11-05 by yoshimalucky