ROBOTIS-GIT/turtlebot3 #331: Turtlebot3_description package missing in melodic?

mkhansen-intel, 2018-10-11:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [x ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [x ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - x[ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - roslaunch turtlebot3_gazebo turtlebot3_world.launch
 
7. Copy and Paste your error message on terminal

    - 
Resource not found: turtlebot3_description
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

  
8. Please, describe detailedly what difficulty you are in 

    - I tried to install from debian but the package isn't found. Is turtlebot3_description supposed to be present in melodic?

mkhansen-intel, 2018-10-16:
Hi @robotpilot - Thank you for the response. I will try to build the turtlebot3_description package from source instead.

closed 2018-10-16 by mkhansen-intel

adauhajre, 2019-01-11:
I just ran into this today building from source running on ubuntu 18 and melodic. @mkhansen-intel did you end up resolving this by building from source?

mylad13, 2022-08-26:
I had the same problem when trying to convert the .xacro turtlebot model to .urdf using the command:
 rosrun xacro xacro -o turtle3burger.urdf turtlebot3_burger.urdf.xacro
where I faced this error:
resource not found: turtlebot3_description
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
I had the turtlebot3_description package installed from source but that still didn't solve the issue. The problem for me was that I had not sourced the catkin_ws! So I just added 
source ~/catkin_ws/devel/setup.bash
to my ~/.bashrc and ran 
source ~/.bashrc
that solved my problem and I was able to run the xacro command.