ROBOTIS-GIT/turtlebot3 #332: Issue with using the turtlebot3_friend/mecanum model

anwaybose, 2018-10-16:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - HERE

anwaybose, 2018-10-16:
Dear Robotis Representative
 I'm building a 4wheel drive mecanum robot using the turtlebot3 waffle_pi kit of ROBOTIS. I've attached two more motors (XL430-W250-T) apart from the two(XM430) that came with the waffle_pi kit. After uploading the code for mecanum to opencr from the example only two motors are working. The motors that came with the waffle_pi (XM430) are working, but the XL430 motors are not working. I've checked the baudrate for both waffle and burger, they are same(57600) but for mecanum it is 1000000. Please help me to solve the issue as soon as possible.
Thanking You
Anway

routiful, 2018-10-16:
Hi :)

First, you are supposed to set same baudrate 57600 or 1000000.
You can change it by Dynamixel-Workbnech or R+ manager.

Which firmware did you used? You made yourself? or provided ones?

anwaybose, 2018-10-17:
The program that I used was the simplest Dynamixel tutorial from the following website (http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/CommandLine).  and I'm getting the folowing error.
[image]

routiful, 2018-10-17:
Single manager only support ONE dynamixel.

Please describe me how to hardware setup.

anwaybose, 2018-10-17:
I've connected one dynamixel to opencr and uploaded the usb_to_dxl code to the opencr. And then I ran the above mentioined firmware which gave me the said error.

routiful, 2018-10-22:
Did you try other baudrate and change ID scan range?

If you did but you can't solve this issue. You can try to use R+ Manager 2.0 to recover firmware of your dynamixel.

http://emanual.robotis.com/docs/en/software/rplus2/manager/#firmware-recovery

closed 2018-10-30 by gpwhd07