ROBOTIS-GIT/turtlebot3 #334: Robot's Actual Linear and Angular Velocities are different than the Command

TheBard2273, 2018-10-24:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?
Raspberry Pi 3

3. Which OS you installed in SBC?
Ubuntu MATE 16.04 or later

4. Which OS you installed in Remote PC?
 Ubuntu 16.04 LTS (Xenial Xerus)
 
Hi,

I have a problem with how my turtlebot3 responds to my commands. I built the robot with a different body but kept the RasPi, OpenCR, LiDAR and Dynamixels. The problem is according to the sketch of the Turtlebot3 Burger which is on the OpenCR, Linear velocity has the unit of (m/s) and angular velocity has the unit of (rad/s) but my robot navigates with different velocities than it's been commanded.
By the way, I have changed the related variables regarding the robot's design in the sketch (WHEEL_RADIUS, WHEEL_SEPARATION, TURNING_RADIUS and ROBOT_RADIUS) yet I still have this problem.
Please Help.
Thanks.

kijongGil, 2018-10-25:
Hi @TheBard2273 :)
I knew about this problem, which will be added in the next update.
But If you want to it now, you can use our [OpenCR-develop][https://github.com/ROBOTIS-GIT/OpenCR/tree/develop].
Change your parameter in this [code](https://github.com/ROBOTIS-GIT/OpenCR/blob/develop/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_burger.h).
Thanks,
Gilbert

TheBard2273, 2018-10-25:
Thank you for your response Gilbert,
I have actually changed those parameters in the sketch that I am uploading on my OpenCR yet I still have the problem. I gave it the new parameters based on the definitions given in the link below on the picture:
 http://emanual.robotis.com/docs/en/platform/turtlebot3/specifications/#data-of-turtlebot3-burger

Is there a specific way in which I have to upload the sketch on OpenCR? Is there a way if I can check to see those parameters actually are changed in my turtlebot ?

TheBard2273, 2018-10-25:
I dug a bit deeper and I think we can get to the bottom of this. If you upload the sketch onto the OpenCR, for getting access to the turtlebot you will have to launch turtlebot3_core.launch file or use "rosrun" and run the rosserial_python node so that system gets calibrated. Now my system gets calibrated but during its process of calibration , in between the log information, right after log info's of subscriber setups I will get the following warning in Yellow :`
"[WARN] [1540467836.069045]: Failed to get param: timeout expired" 
and the the rest of the log info lines show up about TF setups.
Can this be the source of the problem ?

TheBard2273, 2018-10-25:
Let it also be known that the odometry topic understands exactly where robot is and publishes precise data about position of the robot. The problem is that when I ask robot to go on a straight line with velocity of 0.1 m/s for 10 seconds, robot goes 2.6 meters forward instead of one meter.

kijongGil, 2018-10-26:
"[WARN] [1540467836.069045]: Failed to get param: timeout expired" is not related to this problem. This is a parameter used to set the multi-robot. 
I think this problem is not uploaded properly to OpenCR. There is a sendLogMsg() in [turtlebot3_core](https://github.com/ROBOTIS-GIT/OpenCR/blob/185e36be58c4b33bd14d63ac163508aec9ce5fe1/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L707)
You can change log and check upload.

TheBard2273, 2018-10-26:
Can you please elaborate what do you mean by changing the log and checking the upload ?

TheBard2273, 2018-10-26:
I uploaded the original TurtleBot Burger sketch on the OpenCR Board and tested the robot again. It goes forward 2.6 meters again instead of only one meter which I commanded it to go. The only thing that is different now is that the odometry is also wrong and thinks that it has gone forward only one meter.
So uploading the altered sketch will actually have an effect for Odometry Calculation but the motors won't act as they should.

kijongGil, 2018-10-30:
I'm sorry for late reply. 
I mean that for checking upload on the OpenCR your new turtlebot3_core code, modify the print or log line in sendLogMsg().
I tested it with a turtlebot3 with a different wheel and it works well.
If you modify this [line](https://github.com/ROBOTIS-GIT/OpenCR/blob/185e36be58c4b33bd14d63ac163508aec9ce5fe1/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L725), when you run 'turtlebot3_bringup turtlebot3.robot.launch', you can see this log.

TheBard2273, 2018-10-30:
Thank you for the response Gilbert,

I'm sorry if my questions seem a bit dumb but can you help me understand this ?
So if I have a new wheel radius of "0.076 m" and the original turtlebot3 wheel radius is "0.033 m", how exactly should I change the sedLogMsg() ?
I thought this part of the turtlebot3_core code is just for simple Log outputs when you launch the turtlebot3_core.

kijongGil, 2018-11-01:
You can see follow code on 725 line in turtlebot3_core.ino 
> sprintf(log_msg, "Connected to OpenCR board!");

Just change this log to anything, "Connected to your board". 
Then upload on your OpenCR and check this log is print out when you run bringup the robot.

TheBard2273, 2018-11-06:
Sorry for my late reply,
I understand what you're saying, and I have changed that line and yes, I have seen the change being shown in the terminal. So we can understand from this, that the script gets uploaded perfectly.

kijongGil, 2018-11-12:
Sorry for late reply.
Cloud you tell me how to define MAX_LINEAR_VELOCITY and MAX_ANGULAR_VELOCITY in turtlebot3_burger.h?

closed 2018-11-27 by kijongGil