ROBOTIS-GIT/turtlebot3 #338: Dynamixel gets stuck after some movement.

truncs, 2018-10-30:
1. Which TurtleBot3 you have?

    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3

3. Which OS you installed in SBC?
 
    - [x] Raspbian

4. Which OS you installed in Remote PC?
   
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
   
5. Write down software version and firmware version

    - Software version: TurtleBot: ROS kinetic 1.12.13
    - Firmware version: OpenCR 1.3.2
 
6. Write down the commands you used in order

    - Tried resetting the firmware of dynamixel by using R+Manager
    - Then setup the dynamixel using turtlebot_setup_motor
 
  
8. Please, describe detailedly what difficulty you are in 

    - The dynamixel wheels move for some time and then start beeping with USR1 continuously red and after that the wheels don't move at all.

I took a [video](https://www.youtube.com/watch?v=j-P-0-VQsL4) of this if it's helpful.

routiful, 2018-10-31:
Hi :)

Because of low battery, the red led was on. Please check your battery.

truncs, 2018-10-31:
Duh!

closed 2018-10-31 by truncs

routiful, 2018-10-31:
If you have any further issue, please feel free to open issue :)