ROBOTIS-GIT/turtlebot3 #340: Turtlebot3_core doesn't grab the NAMESPACE configuration

brusangel, 2018-10-30:
Hello, I'm working with multiple **turtlebot3**, running this launch file:

**turtlebot3_robot.launch**
```
<launch>
  <arg name="multi_robot_name" default=""/>
  <arg name="set_lidar_frame_id" default="base_scan"/>

  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
    <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
  </include>
  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
    <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
  </include>

  <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
</launch>
```

which launches: **turtlebot3_core.launch**
```
<launch>
  <arg name="multi_robot_name" default=""/>

  <node pkg="rosserial_python" type="serial_node.py" name="turtlebot3_core" output="screen">
    <param name="port" value="/dev/ttyACM0"/>
    <param name="baud" value="115200"/>
    <param name="tf_prefix" value="$(arg multi_robot_name)"/>
  </node>
</launch>
```
The former is being run using this params:
`ROS_NAMESPACE=tb3_1 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:="tb3_1" set_lidar_frame_id:="tb3_1/base_scan"`

but the TF tree doesn't successfully grab the namespace configuration for the `odom` and `base_footprint`: [TF Tree here](https://imgur.com/a/RG4MiWe)

I'm getting this warning on the console when launching (not sure if it has relation or not): [Here](https://imgur.com/a/uYud34B)

I've gone through the source code for both the `serial_node.py` and `SerialClient.py`, and effectively there's no room in the code to grab the namespace for the frames (`odom` & `base_footprint`), I've added the simple fix for the former source file:
`tf_prefix = rospy.get_param('~tf_prefix','')  #line 53`
`client = SerialClient(port_name, baud, tf_prefix=tf_prefix, fix_pyserial_for_test=fix_pyserial_for_test) #line 90`

but for the `SerialClient.py` the patching becomes a bit tricky (posted the same issue at the rosserial_python git but seems like it's not actively maintained, so hope over here I'd get some support)

```
def __init__(self, port=None, baud=57600, tf_prefix=None, timeout=5.0, fix_pyserial_for_test=False): #line 324
# fix for grabbing namespace configurations (line 336)
        if tf_prefix== None:
            # namespace was not initialized
            pass
        else:
            # setting the argument var to class var
            self.tf_prefix=tf_prefix
```
(this was the easy part)

after these come `setupPublisher(self, data):` & `setupSubscriber(self, data):`, and here becomes tricky, I've been spending the last couple of hours working the code, but the debugging gets a bit complex since we're working directly over the robot and the battery runs out very fast. So that's the reason I dare to ask for help from you guys.

(Read the two previous closed issues about this, however it seems like the fix in the firmware wasn't yet added to the mainstream??)

Is it my guess that the problem can be fixed by changing this code correct?, please advice otherwise.

Thanks

ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - Burger
  
2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Raspberry Pi 3
  
3. Which OS you installed in SBC?

    - Ubuntu MATE 16.04 or later
   
4. Which OS you installed in Remote PC?

    - Ubuntu 16.04 LTS (Xenial Xerus)

5. Write down software version and firmware version

    - Software version: [1.12.14]
    - Firmware version: [1.2.3]

brusangel, 2018-10-31:
thanks for your response.

yes, of course, that is implicitly done and understood by default. :)

brusangel, 2018-10-31:
what's the meaning of this warning?
```
[INFO] [1540994241.540698]: Setup subscriber on reset [std_msgs/Empty]
[WARN] [1540994241.771340]: Failed to get param: timeout expired
[INFO] [1540994241.776798]: Setup TF on Odometry [odom]

```
it comes right before the odometry setting up. Can it mean ROS gets a timeout and that's why it can't grab the params to set the `odom` & `base_footprint` frames in order?

routiful, 2018-11-01:
It is time out error caused by rosserial_python, It occationally occured.

brusangel, 2018-11-01:
found the problem in: `turtlebot3_core.ino`, the ROS NodeHandle **nh** is throwing a `timeout` after the method `updateTFprefix()` is being called in `line 80`, specifically in **line 349** `nh.getParam("~tf_prefix", &get_tf_prefix);`, the condition in **line 351** becomes true (due to the timeout) and the frames get "labeled" by the default configuration and not following the `TF prefix` setup...
Would you please help me debug this?, I really need to get over this problem to keep going with my research. Thanks

brusangel, 2018-11-22:
it seems under some wifi environments if the AP is with traffic (means an AP deployed under a normal environment but testing), the latency on the network introduces a delay on the ROS messages, and the ends cannot grab the params before the timeout is reached. What I did to solve the issue was to increase both the buffer size and the timeout limit, recompiled and the problem was solved.

wx0205, 2018-12-20:
I'm working on navigation with multi-turtlebot3. When I launch them in different namespace, the turtlebot3 can't navigete. And I give initial pose in the Rviz, it doesn't work.  How can i solve it?
Of cource, I want three turtlebot3 in the same rviz. I can give different targets, and  the turtlebot3 can navigate independently. It's diffcult for me. Can you give me some advice? Thank you very much!

routiful, 2018-12-21:
@wx0205  Please open another issue. Your comment is not proper in this thread.

gpwhd07, 2019-01-29:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-01-29 by gpwhd07

EndlessLoops, 2019-12-06:
@brusangel @routiful 
Hi，guys
  I got the seem problem.I tried to increase the value of timeout in SerialClient.py file and the value of buffer_size based on [this](https://github.com/ROBOTIS-GIT/turtlebot3/issues/160#issuecomment-514896184).But it can not work.

`[INFO] [1575524884.421414]: Setup subscriber on reset [std_msgs/Empty]
[WARN] [1575524884.450769]: Failed to get param: timeout expired
[INFO] [1575524884.455505]: Setup TF on Odometry [odom]
[INFO] [1575524884.459867]: Setup TF on IMU [imu_link]
[INFO] [1575524884.464465]: Setup TF on MagneticField [mag_link]
[INFO] [1575524884.468976]: Setup TF on JointState [base_link]
[INFO] [1575524884.493382]: --------------------------
[INFO] [1575524884.497909]: Connected to OpenCR board!
[INFO] [1575524884.502435]: This core(v1.2.3) is compatible with TB3 Burger`

This machine was probably bought at the end of 2017.
How did you solve this problem? Can you tell me a detailed solution to this problem?