ROBOTIS-GIT/turtlebot3 #343: Convert a Turtlebot3 Burger to Waffle

stevensu1838, 2018-11-05:
Hi team,

I am converting a Turtlebot3 into a Waffle. I've noticed that the main difference is the servo motors. Do I need to change the motors? Or I can just use the motors of Burger? Do I need to change anything in the packages? Thanks a lot.

kijongGil, 2018-11-06:
Hi @stevensu1838 :)
Yes, you can change DXL. You don't need to change anything. However, If you change wheel size, you have to change OpenCR frimware. If you want to change wheel size, refer to [here](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_waffle/turtlebot3_core/turtlebot3_waffle.h).

Thanks,
Gilbert

stevensu1838, 2018-11-09:
Hi buddy,
If I keep using the motors of burgers, can you please tell me where and what I should modify to make it work well? Also, this version of Realsense is out of date and hard to find online now and I need to do Octomap with the 3D camera, do you know any others which work well with turtlebot3? Thanks a lot

kijongGil, 2018-11-27:
I'm sorry for late reply. you have to set XL430 as like XM430. please check DLX's ID, baudrate and protocol version. You can check and modify it using [R+ manager](http://emanual.robotis.com/docs/en/software/rplus2/manager/).
We keep using the realsense.

gpwhd07, 2018-12-12:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-12-12 by gpwhd07

stevensu1838, 2019-01-14:
Hi @kijongGil ,

I am also planning to put on a NUC on my turtlebot. Can I please ask how you power your NUC(16 V required) on the robot? Also, does NUC work well with Opencr via USB connection(NUC features USB 3.0 and Opencr has USB 2.0)? Much appreciated if you could post a pic of your component connections. Thanks a lot.