ROBOTIS-GIT/turtlebot3 #344: No Transform from [Camera] to frame [base_footprint]

ilhamasy, 2018-11-05:
[image]
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [V] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [V] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [V] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [V] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)
 
5. Write down the commands you used in order

    - $ export TURTLEBOT3_MODEL=burger
      $ roslaunch turtlebot3_automatic_parking_vision turtlebot3_automatic_parking_vision.launch
 
6. Copy and Paste your error message on terminal

    - Could not find a connection between 'base_footprint' and 'camera' because they are not part of the same tree.Tf has two or more unconnected trees.

  
7. Please, describe detailedly what difficulty you are in 

    - I was follow tutorial by emanual robotis turtlebot3, automatic parking vision. Camera calibration no one warning everything going well, and rviz show up, and finally my turtlebot just spinning around, and the command was error. i checked on rviz in the section TF subsection 'camera' : No transform form [camera]  to frame [base_footprint]
and camera_rgb_optical_frame : No transform form [camera_rgb_optical_frame] to frame [base_footprint]

Rostopic List:
[image]

$roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch 
[image]

[image]

kijongGil, 2018-11-06:
Hi @ilhamasy :)
Burger has not camera. The turtlebot3_automatic_parking_vision is for waffle_pi.
If you want to use this package, please `export TURTLEBOT3_MODEL=waffle_pi`

Thanks,
Gilbert

ilhamasy, 2018-11-06:
Thanks gilbert for response :)

Can I use Turtlebot Burger for this application automatic parking vision
that should be used on turtlebot waffles?

kijongGil, 2018-11-07:
It is not impossible. However, you have to make tf of camera in urdf as like waffle_pi.

gpwhd07, 2018-11-15:
This issue will be closed since there were no actions for a while. 
Please try to make TF of camera in urdf about  this application automatic parking vision.
If you have more question, you can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-15 by gpwhd07