ROBOTIS-GIT/turtlebot3 #345: Update needed on teleportation with Leap Motion

stevensu1838, 2018-11-07:
In the teleoperation part of user manual, you have the following instructions: 
LEAP Motion [Remote PC] Connect LEAP motion to the remote PC via Bluetooth.  [Remote PC] Install packages for teleoperation using LEAP motion.  

Can you please show me more details on how to accomplish this task? I think we need a package to build up the communication between Leap Motion and turtlebot3? But I don't see it anywhere in the github page. Can you please update it?

stevensu1838, 2018-11-18:
Hi buddy, 
I am on it and will show you my own node soon. Just a quick question. Do you know how I can get a better understanfing of ROS leap diver? Before you told me, I didn't even know that I need to use leap_hands.cpp file and subscribe to marker. How did you find out that leap_hands.cpp is the file we need to use? Why not the python files in script folder? How can I learn more about marker? Thanks with no end.

stevensu1838, 2018-11-18:
Btw, can you please show me the difference markers and markers arrays messages? I think I am going to figure it out first. Thanks

stevensu1838, 2018-11-25:
Hi man, 
i   don't think I am capable of making this node. coz I don't really understand the leapros pkg. Can any of you experts give me a favor? since you've done this before. I simply want to teleoperate my turtlebot 3 as you showed in your manual which is not working now. Would you consider updating it in your e-manual?
Cheers

routiful, 2019-01-29:
I assume that you found a solution to work turtlebot3 with Leap Motion.
I will close this issue. If you have any further questions please feel free to open it.

closed 2019-01-29 by routiful