ROBOTIS-GIT/turtlebot3 #349: Navigation in Gazebo not working with realistic update rate of laser scanner

shahin-y, 2018-11-22:
Hi,

I try to run the Turtlebot3 navigation demo in Gazebo as described in the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#simulation).
However, with the normal configuration of the Turtlebot3 Bruger Gazebo model I always run into the following problem:
[image]
However, if I change from `<update_rate>5</update_rate>` to a higher value like 25 Hz this issue does not arise. Probably this didn't occur with the update rate at 1800 Hz as well which was changed in #258. As 5 Hz is the frequency of the real sensor it should run with this value. Do you run into this problem as well?

It should be reproducible by following the steps in the e-manual:
1. Launch simulation:  'export TURTLEBOT3_MODEL=burger' then  'roslaunch turtlebot3_gazebo turtlebot3_world.launch'
2. Launch navigation: 'export TURTLEBOT3_MODEL=waffle_pi' then 'roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=/home/some_dir/catkin_ws/src/turtlebot3/turtlebot3_navigation/maps/map.yaml' (If you installed from source specify the file location accordingly)

My setup:
ROS Melodic, Gazebo version 9.0.0 on Ubuntu 18.04
I installed this (meta)package from source.
The only changes I made were in fixing the map.yaml (like: #348) and removing the leading slashes from the frames specified in the [global](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/global_costmap_params.yaml) and [local costmap](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/local_costmap_params.yaml) parameter files. As the ROS navigation stack switched to tf2 we need to remove the leading slashes to comply with tf2.

Do you encounter this problem as well? Any help appreciated, thanks!

kijongGil, 2018-11-23:
Hi @shahin-y :)
Unfortunately, we not support ROS Melodic, Gazebo version 9.0.0 on Ubuntu 18.04 yet.
We support ROS Kinetic, gazebo 8 and Ubuntu 16.04. How about change your environments? Or please wait for our new release about ROS melodic and ubuntu 18.04 in next year.

Thanks,
Gilbert

kijongGil, 2018-11-27:
I'm sorry for late reply. We have not been ready to release yet. So, If we'll encounter this issue, let's talk again.

kijongGil, 2019-01-02:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-01-02 by kijongGil

jkeane508, 2022-03-21:
> Hi @shahin-y :) Unfortunately, we not support ROS Melodic, Gazebo version 9.0.0 on Ubuntu 18.04 yet. We support ROS Kinetic, gazebo 8 and Ubuntu 16.04. How about change your environments? Or please wait for our new release about ROS melodic and ubuntu 18.04 in next year.
> 
> Thanks, Gilbert

thanks fart face