ROBOTIS-GIT/turtlebot3 #351: Turtlebot don't move

quip95, 2018-12-04:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [+] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [+] INTEL NUC 

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [+] Ubuntu 16.04 LTS (Xenial Xerus)

4. Which OS you installed in Remote PC?

    - [+] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [1.12.14]
    - Firmware version: [1.3.2]
 
6. Write down the commands you used in order

-  roslaunch turtlebot3_bringup turtlebot3_robot.launch
started roslaunch server http://localhost:43983/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

auto-starting new master
process[master]: started with pid [30950]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f4bc6910-f7c0-11e8-94ed-88b11116d433
process[rosout-1]: started with pid [30963]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [30967]
[INFO] [1543926876.294996]: ROS Serial Python Node
[INFO] [1543926876.309214]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1543926880.786663]: Note: publish buffer size is 1024 bytes
[INFO] [1543926880.787324]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1543926880.791211]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1543926880.987357]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1543926880.995942]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1543926881.074045]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1543926881.081365]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1543926881.088756]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1543926881.096880]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1543926882.814370]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1543926882.822962]: Note: subscribe buffer size is 1024 bytes
[INFO] [1543926882.824044]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1543926882.838255]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1543926882.846524]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1543926882.852195]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1543926883.824171]: Setup TF on Odometry [odom]
[INFO] [1543926883.826076]: Setup TF on IMU [imu_link]
[INFO] [1543926883.827877]: Setup TF on MagneticField [mag_link]
[INFO] [1543926883.829590]: Setup TF on JointState [base_link]
[INFO] [1543926883.834018]: --------------------------
[INFO] [1543926883.835810]: Connected to OpenCR board!

- roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

- upload turtlebot3_setup_motor

opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_869990/turtlebot3_setup_motor.ino.bin 
file size : 115 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 1.097000 sec
flash_write : 0 : 1.923000 sec 
CRC OK B6ACB6 B6ACB6 0.004000 sec
[OK] Download 
jump_to_fw 

Succeeded to open the port!
 - Device Name : /dev/OpenCR
 - Baudrate    : 57600

Start turtlebot3 setup motor
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> Do you really want to setup ? y/n : yes
setup.... left
Find Motor...
    setbaud : 57600
[TxRxResult] There is no status packet!
    setbaud : 1000000
[TxRxResult] Incorrect status packet!
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> Do you really want to setup ? y/n : yes
setup.... right
Find Motor...
    setbaud : 57600
[TxRxResult] There is no status packet!
    setbaud : 1000000
[TxRxResult] Incorrect status packet!
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> test.... left
Test Motor Left...
    no dxl motors
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> test.... right
Test Motor Left...
    no dxl motors
 
- Upload DynamixelWorkbench monitor 

[CMD] : begin 57600
Succeed to begin(57600)
[CMD] : scan
Can't find Dynamixel
[CMD] : begin 1000000
Succeed to begin(1000000)
[CMD] : scan
Can't find Dynamixel

8. Please, describe detailedly what difficulty you are in 

Hi!

For several days after replacing the OpencCR on the other OpenCR, the motors don't work.
Dynamixel motors don't move when I send velocity commands or run roslaunch teleop.
 I get errors when setting the ID in the Arduino IDE. 

Thanks

kijongGil, 2018-12-06:
Hi @ quip95 :)
Did you update your OpenCR replaced? If you are not, refer to [here](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-firmware-upload-for-tb3).

quip95, 2018-12-06:
@kijongGi , Hi, thanks for reply. 
Yes, I successfully updated OpenCR before testing.

kijongGil, 2019-01-02:
I'm sorry for late reply.
How about your problem?

stevensu1838, 2019-01-14:
Hi @quip95 ,

I am also planning to put on a NUC on my turtlebot. Can I please ask how you power your NUC(16 V required) on the robot? Also, does NUC work well with Opencr via USB connection(NUC features USB 3.0 and Opencr has USB 2.0)? Much appreciated if you could post a pic of your component connections. Thanks a lot.

quip95, 2019-01-14:
@kijongGil , I don't use turtlebot3_setup_motor anymore and everything works well.  
But when I run turtlebot3_setup_motor in the Arduino IDE an error  "There is no status packet!" occurs.

quip95, 2019-01-14:
@stevensu1838 ,  For now we use the power network for charging NUC (19V)  and 11V battery for OpenCR.
NUC work well with OpenCR via USB.

[image]

kijongGil, 2019-01-17:
@quip95, did you turn on the OpenCR? because in picture you posted OpenCR is turn off.

closed 2019-01-24 by kijongGil