ROBOTIS-GIT/turtlebot3 #352: No map received in Rviz using Turtlebot3 in Gazebo

samialperen, 2018-12-04:
Hello guys,

I do not have any Turtlebot 3 ,but i am trying to simulate Turtlebot 3 on Gazebo and use gmapping for SLAM. I followed SLAM tutorial like this:

```
$ export TURTLEBOT3_MODEL=waffle_pi
$ roslaunch turtlebot3_gazebo turtlebot3_world.launch
$ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
```

There is no problem until here. I can move turtlebot3. I can also see odometry of the robot and laser scan datas in rviz. However, i can not see any map produced by gmapping. In below you can see some results:
[image]
                                                          Figure 1: Rviz output
[image]
                                                           Figure 2: rqt_graph output

Can somebody help me what am i missing? Thank you very much!

kijongGil, 2018-12-06:
Hi @samialperen :)
In Fig 1, you can see the map in display panel. please show me the map information?

samialperen, 2018-12-06:
Hey, it states no map received in Rviz display panel.

yoshimalucky, 2018-12-19:
@samialperen Open the TF tab and show us.

Otherwise, run this command and attach the image of the tree in the PDF file:
`rosrun tf view_frames`

yoshimalucky, 2018-12-19:
@samialperen First of all, which binary or source do you use?

If you use binary, just update the packages:
```
sudo apt-get update
sudo apt-get upgrade
```
Then, you will make it. I made it by this.

For source, show us the log of:
```
roscd turtlebot3
roscd git show
```

kijongGil, 2019-01-02:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-01-25 by kijongGil

nagarjunvinukonda, 2021-05-13:
Hi, I am facing the same problem. can this issue be reopened?

I am using ROS melodic 18.0.4.

I am trying to gmap the turtlebot3_world.launch file using the command:
`roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping 
`
But, I am having trouble with **No map received:**

As  @routiful  told above I dont find any mistake in my terminal while I launch gmapping, T**His is the following output in my terminal :**

`
nagarjunv@nagarjunv-Inspiron-7580:~/hk_ws$ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping 
... logging to /home/nagarjunv/.ros/log/891ec6d0-b42e-11eb-90d9-3c6aa7417fe4/roslaunch-nagarjunv-Inspiron-7580-12859.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://nagarjunv-Inspiron-7580:38227/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /turtlebot3_slam_gmapping/angularUpdate: 0.2
 * /turtlebot3_slam_gmapping/astep: 0.05
 * /turtlebot3_slam_gmapping/base_frame: base_footprint
 * /turtlebot3_slam_gmapping/delta: 0.05
 * /turtlebot3_slam_gmapping/iterations: 5
 * /turtlebot3_slam_gmapping/kernelSize: 1
 * /turtlebot3_slam_gmapping/lasamplerange: 0.005
 * /turtlebot3_slam_gmapping/lasamplestep: 0.005
 * /turtlebot3_slam_gmapping/linearUpdate: 1.0
 * /turtlebot3_slam_gmapping/llsamplerange: 0.01
 * /turtlebot3_slam_gmapping/llsamplestep: 0.01
 * /turtlebot3_slam_gmapping/lsigma: 0.075
 * /turtlebot3_slam_gmapping/lskip: 0
 * /turtlebot3_slam_gmapping/lstep: 0.05
 * /turtlebot3_slam_gmapping/map_frame: map
 * /turtlebot3_slam_gmapping/map_update_interval: 2.0
 * /turtlebot3_slam_gmapping/maxUrange: 3.0
 * /turtlebot3_slam_gmapping/minimumScore: 50
 * /turtlebot3_slam_gmapping/odom_frame: odom
 * /turtlebot3_slam_gmapping/ogain: 3.0
 * /turtlebot3_slam_gmapping/particles: 100
 * /turtlebot3_slam_gmapping/resampleThreshold: 0.5
 * /turtlebot3_slam_gmapping/sigma: 0.05
 * /turtlebot3_slam_gmapping/srr: 0.1
 * /turtlebot3_slam_gmapping/srt: 0.2
 * /turtlebot3_slam_gmapping/str: 0.1
 * /turtlebot3_slam_gmapping/stt: 0.2
 * /turtlebot3_slam_gmapping/temporalUpdate: 0.5
 * /turtlebot3_slam_gmapping/xmax: 10.0
 * /turtlebot3_slam_gmapping/xmin: -10.0
 * /turtlebot3_slam_gmapping/ymax: 10.0
 * /turtlebot3_slam_gmapping/ymin: -10.0

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    turtlebot3_slam_gmapping (gmapping/slam_gmapping)

ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [12894]
process[turtlebot3_slam_gmapping-2]: started with pid [12895]
process[rviz-3]: started with pid [12896]

`
The following are the images from Rviz:
[image]

I have downloaded Turtlebot Simulations and turtlebot3 packages from the following link:

https://github.com/ROBOTIS-GIT/turtlebot3_simulations
https://github.com/ROBOTIS-GIT/turtlebot3

I have freshly downloaded the above packages and catkin_make. But still don't work while doing mapping.

The following is image of rqt_graph:

[image]

THe following is image of tf_tree:

[image]

Can anyone please let me know what exactly is the problem and how to solve it?