ROBOTIS-GIT/turtlebot3 #353: Dynamixel messages are getting switched with ID while recording a bag

anwaybose, 2018-12-05:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ 1] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ 1] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ 1] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [1 ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 
I am using waffle-pi to make a 4wheel mecanum drive robot, and have written ros messages to publish the encoder data. I am able to get those datas published in ROS, but sometimes the encoder values are getting switched with ID, which in return recording incorrect data. But sometimes it is recording correctly the datas corresponding to the respective wheel-ids. What can be the reason behind this???

kijongGil, 2018-12-06:
Hi @anwaybose :)
I don't know about how you made ros messages to publish the encoder data and how you get the encoder values. Did you modify OpenCR code for mecanum drive robot ?

anwaybose, 2018-12-06:
Yes, obviously I have modified the opencr code for mecanum to publish messages in ROS and have made the necessary .msg and .h files.

routiful, 2018-12-12:
Hi, did you check the SyncRead function is working properly by Serial.print?

anwaybose, 2018-12-12:
If 
bool Turtlebot3MotorDriver::readEncoder(int32_t &left_rear_value, int32_t &right_rear_value, int32_t &left_front_value, int32_t &right_front_value)
{
  int dxl_comm_result = COMM_TX_FAIL;              // Communication result
  bool dxl_addparam_result = false;                // addParam result
  bool dxl_getdata_result = false;                 // GetParam result

  // Set parameter
  dxl_addparam_result = groupSyncReadEncoder_->addParam(left_rear_wheel_id_);
  Serial.print("Wheel ID LRW:");Serial.println(left_rear_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    return false;
////  
  dxl_addparam_result = groupSyncReadEncoder_->addParam(right_rear_wheel_id_);
  Serial.print("Wheel ID RRW:");Serial.println(right_rear_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    return false;
//
  dxl_addparam_result = groupSyncReadEncoder_->addParam(left_front_wheel_id_);
  Serial.print("Wheel ID LFW:");Serial.println(left_front_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
   return false;
//
  dxl_addparam_result = groupSyncReadEncoder_->addParam(right_front_wheel_id_);
  Serial.print("Wheel ID RFW:");Serial.println(right_front_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    return false;

this was the code then it was getting stuck in the loop. So I have changed it to

bool Turtlebot3MotorDriver::readEncoder(int32_t &left_rear_value, int32_t &right_rear_value, int32_t &left_front_value, int32_t &right_front_value)
{
  int dxl_comm_result = COMM_TX_FAIL;              // Communication result
  bool dxl_addparam_result = false;                // addParam result
  bool dxl_getdata_result = false;                 // GetParam result

  // Set parameter
  dxl_addparam_result = groupSyncReadEncoder_->addParam(left_rear_wheel_id_);
  Serial.print("Wheel ID LRW:");Serial.println(left_rear_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    groupSyncReadEncoder_->clearParam();
    dxl_addparam_result = groupSyncReadEncoder_->addParam(left_rear_wheel_id_);
    Serial.print("Wheel ID LRW:");Serial.println(left_rear_wheel_id_);
    Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    return false;
////  
  dxl_addparam_result = groupSyncReadEncoder_->addParam(right_rear_wheel_id_);
  Serial.print("Wheel ID RRW:");Serial.println(right_rear_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    return false;
//
  dxl_addparam_result = groupSyncReadEncoder_->addParam(left_front_wheel_id_);
  Serial.print("Wheel ID LFW:");Serial.println(left_front_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
   return false;
//
  dxl_addparam_result = groupSyncReadEncoder_->addParam(right_front_wheel_id_);
  Serial.print("Wheel ID RFW:");Serial.println(right_front_wheel_id_);
  Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);
  if (dxl_addparam_result != true)
    return false;

 I thought that parameters were not cleared probably so it was returning false. That's why I have added the part  
 if (dxl_addparam_result != true)
    groupSyncReadEncoder_->clearParam();
    dxl_addparam_result = groupSyncReadEncoder_->addParam(left_rear_wheel_id_);
    Serial.print("Wheel ID LRW:");Serial.println(left_rear_wheel_id_);
    Serial.print("dxl_addparam_result :");Serial.println(dxl_addparam_result);

And then it was not getting stuck in the loop and the motors were moving.
And values were getting printed by Serial.print

anwaybose, 2018-12-13:
Thanks @routiful for your suggestion.
But, can you tell me what's the reason behind switching of encoder messages with DXl ids in irregular manner??? That is the main issue.

routiful, 2018-12-19:
the Dynamixel ID have used two functions (readEncoder() and controlMotor())
readEncoder will be called in loop and controlMotor will be called timer.
I think this process makes happen to access simultanously Dynamixel ID.

Can you try to move controlMotor function to loop? Like below

```
  if ((t-tTime[1]) >= (1000 / DRIVE_INFORMATION_PUBLISH_FREQUENCY))
  {
    controlMotor()
    tTime[1] = t;
  }
```

closed 2018-12-26 by gpwhd07