ROBOTIS-GIT/turtlebot3 #354: WAFFLE_PI cannot tf and map properly

young12345678, 2018-12-07:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [Y ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ Y] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [Y ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [Y ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    [REMOTE PC]
    -roscore

    [TURTLEBOT]
   - source ~/catkin_ws/devel/setup.bash
   - roslaunch turtlebot3_bringup turtlebot3_robot.launch
   
   [REMOTE PC]
    - source ~/catkin_ws/install_isolated/setup.bash
    - export TURTLEBOT3_MODEL=waffle_pi
    - roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping

7. Copy and Paste your error message on terminal

    - 
[image]

  
8. Please, describe detailedly what difficulty you are in 

    - when I try to use gmapping to map, the waffle_pi cannot move properly in the rviz and the rviz shows a status error as follow:
    base_footprint No transform from [base_footprint] to [map]
    base_link          No transform from [base_link] to [map]
    base_scan        No transform from [base_scan] to [map]
    camera_link       No transform from [camera_link] to [map]
     ...
     ...
     ...

kijongGil, 2018-12-11:
Hi @young12345678 :)
This problem is that turtlebot3 was not connected to PC.
Did you set the MASTER IP and turtlebot IP?
If you are not, refer to [here](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#network-configuration).

Thanks,
Gilbert

young12345678, 2018-12-12:
> Hi @young12345678 :)
> This problem is that turtlebot3 was not connected to PC.
> Did you set the MASTER IP and turtlebot IP?
> If you are not, refer to [here](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#network-configuration).
> 
> Thanks,
> Gilbert

Thank you, but actually, I have made the turtlebot3 connect to the PC, and the ridar works fine in the rviz of the remote PC.  The problem is that the transform process failed in the rviz.

routiful, 2018-12-13:
Did you try timesync between PC and RP3?

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

young12345678, 2018-12-17:
> Did you try timesync between PC and RP3?
> 
> http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

Oh, thank you so much, time is the problem. Because i use local area network to connect waffle_pi, every time when i turn on the Raspberry Pi，the time is not synchronized with the remote PC. Then i use ntpdate to synchronization, it works!

closed 2018-12-17 by young12345678

routiful, 2018-12-17:
Great :) If you have any further question, please feel free to open new ticket.