ROBOTIS-GIT/turtlebot3 #355: Question about TurtleBot3 navigation

HubertRoginski, 2018-12-07:
I work with a robot(Turtlebot3 burger) and I can not find anywhere about which algorithm is used by default in navigation global planner. I know that local planner is dynamic window approach(dwa_local_planner). The only information about global planner I found was the algorithm Rapidly Exploring Random Trees (RRT), but I do not know if this is the default setting. I will add that I did not change anything in the robot settings, I use official packages. I am asking for clarification of the problem and information about the navigation algorithms used. I generally need information on what happens in the program when I set the "2D Nav Goal" in RVIZ during navigation.

kijongGil, 2018-12-11:
Hi, @HuberRoginski :)
Turtlebot3 navigation use [move_base](http://wiki.ros.org/move_base) and [amcl](http://wiki.ros.org/amcl) package. Local planner and global planner are in move_base. 
Thanks,
Gilbert

HubertRoginski, 2018-12-12:
Hi, @kijongGil :)
In my turtlebot3_navigation-> move_base.launch i have only base_local_planner.
[image]
Based on what you said, I checked the move_base package, there are parameters:
~base_global_planner (string, default: "navfn/NavfnROS" For 1.1+ series)
~base_local_planner (string, default: "base_local_planner/TrajectoryPlannerROS" For 1.1+ series)

But I still do not know what algorithm is used, A*, Dijkstra's, High-speed Navigation Using the
Global Dynamic Window Approach, Rapidly Exploring Random Trees (RRT)?
Can you just write what algorithm is used in Turtlebot3 for the local planner and global planner?

routiful, 2018-12-13:
The default planner in TurtleBot3 is Dynamic Window Approach Planner (http://wiki.ros.org/dwa_local_planner)

HubertRoginski, 2018-12-26:
I found in the book "ROS Navigation Tuning Guide" Kaiyu Zheng September 2, 2016 this:
[image]
Link: http://cs.stanford.edu/group/manips/publications/pdfs/Brock_1999_ICRA.pdf
This is about "High-speed Navigation Using the Global Dynamic Window Approach "
Everything you said is true, but can anyone explain to me in a simple way what Global Dynamic Window Approach is responsible for, and what Djikstra's algorithm is for.(during navigation)

KimHyung, 2019-06-05:
global planner (navfn/NavfnROS) uses Dijkstra's algorithm.
local planner uses DWA's algorithm.

HubertRoginski, 2019-06-05:
Thanks, that is the best answer.

aliozcan01, 2020-05-03:
> global planner (navfn/NavfnROS) uses Dijkstra's algorithm.
> local planner uses DWA's algorithm.

Best answer... Thank you !

palwanishan, 2021-08-13:
How can I use another algorithms for global and local planners?

tessyninan, 2025-01-16:
is it possible to use other algorithms for pathplanning and adding manipulator to turtlebot3 burger?