ROBOTIS-GIT/turtlebot3 #356: Intel RealSens R200 camera not available on line any more. Any other 3D cameras can replace it?

stevensu1838, 2018-12-12:
Hi,
I am buying an Intel RealSens R200 camera for my Turtlebot 3  but can't find it anywhere online. Can you help? If not, is there any other 3D camera you've tested that can replace Intel RealSens R200?

How about Intel SR300, Intel ZR300, Intel Euclid, Asus?
Thanks

Specifically, I wanna build Octomap with the camera.

routiful, 2018-12-13:
I heard that Realsens R200 is not sale anymore.
You can find another camera in [intel store](https://click.intel.com/realsense.html)

yoshimalucky, 2018-12-19:
D435i is the best!
https://click.intel.com/intel-realsense-depth-camera-d435i-imu.html

stevensu1838, 2018-12-24:
@yoshimalucky
Hi buddy, do you have any documents or tutorial on how to use D435i in ROS? Thanks a lot

kijongGil, 2019-01-25:
Sorry. We don't have experiment with D435i.

closed 2019-01-25 by kijongGil